autorope/donkeycar #14: Improve data augmentation with shifted images

wroscoe, 2016-12-30:
Here is a great example: https://chatbotslife.com/using-augmentation-to-mimic-human-driving-496b569760a9#.b6jyrypqu

alanswx, 2017-02-13:
I don't know if this improves things, or if this augmentation is quite right, but you can do a lot of it with one call:

             X_train, X_validation, y_train, y_validation = train_test_split(X, Y, test_size=0.20, random_state=42)
             datagen = ImageDataGenerator(
                rotation_range=1,
                    width_shift_range=0.2,
                    height_shift_range=0.2,
                    shear_range=0.2,
                    zoom_range=0.02,
                    horizontal_flip=False)

             # compute quantities required for featurewise normalization
             # (std, mean, and principal components if ZCA whitening is applied)
             datagen.fit(X_train)

             # fits the model on batches with real-time data augmentation:
             m.fit_generator(datagen.flow(X_train, y_train, batch_size=32, shuffle=True),
                    samples_per_epoch=len(X_train)*3, nb_epoch=20,callbacks=callbacks_list,
                    validation_data=(X_validation, y_validation)
                    )

wroscoe, 2017-02-13:
Parts of that will work. We'll want to validate on non-augmented data.

alanswx, 2017-02-13:
That snippet does validate on non-augmented data. (I believe)

wroscoe, 2017-02-17:
You're right @alanswx but it doesn't switch the steering angles when you flip the image horizontally.

alanswx, 2017-02-17:
that is why i turned off the flipping code -- once i get my car driving i will check in a class that does all the augmentation

wroscoe, 2018-11-27:
I've moved the augmentation code to the donkeybarn repo. https://github.com/autorope/donkeybarn/blob/master/donkeybarn/cv/augment.py

closed 2018-11-27 by wroscoe