autorope/donkeycar #23: Update build instructions.

wroscoe, 2017-01-25:
Include
* How to pick a car + tested models. 
* link to ami image
* Wiring diagram + pictures

yconst, 2017-01-27:
Are the current build instructions up to date? Installing to a RPi2 via the docs at https://github.com/wroscoe/donkey/blob/master/docs/01-build_a_car.md and then trying to run drive_pi.py shows an error about missing docopt. Any ideas?

wroscoe, 2017-01-27:
docopt should be installed when you run `pip install -e ../donkey[pi]` in your virtual environment.

That error should be fixable by installing docopt  in your virtual environment
`pip install docopt`

I'm updating the build instructions to be clearer but the current version should work.

wroscoe, 2017-01-27:
@yconst  Here is the in progress build doc. Please suggest update and add pictures. https://docs.google.com/document/d/11IPqZcDcLTd2mtYaR5ONpDxFgL9Y1nMNTDvEarST8Wk/edit#

yconst, 2017-01-27:
Thanks for sharing! I'll sure chime in if I have any suggestions, although I'm still getting to grips with the entirety of the project!

alanwells, 2017-02-07:
@wroscoe Thanks for this great project! I've got my car built up (based on an [ExceedRC Magnet 1/16 scale truck)](http://www.nitrorcx.com/51c853-savared-24-ghz.html) and steering mostly works (angles are a little off wheels move left/right), but throttle doesn't work at all. The error code on the ESC indicates abnormal throttle signal, so I'm thinking it's a PWM calibration issue. I noticed there was some stubbed out calibration code in the `actuators.py`, do you have any suggestions on where to start on figuring out the correct values? 

Also, I have a few notes from going thru the donkey install process on car and server. Do you prefer to receive them as suggestions to the Google Doc or in comments here?

Here are a few pics of the build. The 1/16th scale is a bit cramped but it worked out OK. Happy to take better pics for the build document if it's helpful:

[image]

[image]

wroscoe, 2017-02-07:
That car looks awesome. 

For doc suggestions please make them in the Google Doc. https://docs.google.com/document/d/11IPqZcDcLTd2mtYaR5ONpDxFgL9Y1nMNTDvEarST8Wk/edit#

@adammconway will know more about ESC settings but i'd using script that loops through all possible ESC values to find the correct range. If the motor is brushed it will likely need to be calibrated by setting full, 0, reverse throttle values. before using.

alanwells, 2017-02-08:
@wroscoe @adammconway got the throttle calibration issue sorted out. I was making the mistake of turning on the ESC before I had the pilot running on the pi, so the ESC couldn't self-calibrate. I added the Exceed Magnet to the tested vehicle list on the google doc.

closed 2017-02-09 by wroscoe