autorope/donkeycar #27: Refactor imports?

yconst, 2017-01-28:
Donkey is importing all modules in the donkey/ directory, which leads to a lot of unneeded module imports, and breaks in many cases (such as running drive_pi.py, which requires envoy and Keras, even thought they are not used). Can this be architected differently so that only modules needed are imported?

wroscoe, 2017-01-28:
Yes. I've also had issues with this. There's some balance between having modules accessible after `import donkey as dk`` and not importing modules that won't get used or cause errors on certain platforms. This could be changed in   https://github.com/wroscoe/donkey/blob/master/donkey/__init__.py

yconst, 2017-01-30:
init checks for os.uname()[1], which gives the node name and can differ from machine to machine. Maybe it's better to check with something like: `os.uname()[4].startsWith("arm")` (per [this SE question](http://raspberrypi.stackexchange.com/questions/5100/detect-that-a-python-program-is-running-on-the-pi))?

PR #29 (2017-02-09): Add differential drive and update docs.
* Creates an abstraction of mixers over actuators to support differential drive vehicles (untested).
* Moves pilot and session control into the web server.
* Adds web interface to switch pilots
* Adds web interface to view sessions and delete photos.
* Fixes joystick so full throttle is possible at sharp angles.
* Moves most documentation to a google doc
* Rename demos folder to scripts
* Add line detection implementation of CompoundEye using opencv

Issues
fixes #27 #23 #32 #21 #20 #19 #1
closed 2017-02-09 by wroscoe