autorope/donkeycar #54: Allow RC control of car (for training or driving around)

yconst, 2017-02-28:
As I understand RemoteClient already sends angle and throttle data when making the request to the Tornado server. I guess it would be feasible to get these values from an RC receiver (whose driver needs implementing), and pass them to the server.

wroscoe, 2017-02-28:
@yconst Yes this is feasible but it would add hardware and software complexity. Other car platforms I've seen use an arduino between the receiver and ESC to sniff the values and then record them. This also gets tricky for the platform to support user controls coming from the car and the web. Given that the web control can make car updates around ~30fps on a local network and since we now have support for Play Station controls through the browser what's the advantage you see to using the RC control?

yconst, 2017-02-28:
I wasn't aware of the support for Playstation controls. That sounds great. Can you share some details on browser support and how to get it to work?

Re. RC, you are right that an interface is required, either in the form of an Arduino or otherwise. To be frank, I had in mind the NAVIO HAT, where RC input decoding is built in, so it adds minimal complexity.

wroscoe, 2017-02-28:
No docs yet but here's an email dump of the contributor (Alan Steremberg)  explaining it.

I turned it on in chrome:
chrome://flags/#enable-gamepad-extensions

Here is my checkin:
https://github.com/alanswx/donkey/commit/d49a813976fd4e010bfe87a165166b294db4ba28

It is setup to use the right stick on my playstation 3 controller.. I plugged it in via USB to my laptop. I didn't bother dealing with bluetooth yet, it doesn't seem super important.

It seems to drive really well - I can't wait to get some training data (unfortunately there are cars parked in the garage on my track!!)

yconst, 2017-02-28:
Great stuff! I'll be testing this for sure.

closed 2017-03-02 by wroscoe