autorope/donkeycar #57: Implement odometry with hall effect sensor

alanwells, 2017-03-02:
The plan is to attempt an odometer setup inspired by [this post](https://sites.google.com/site/wayneholder/self-driving-rc-car/mechanical-odometry-and-speed-control).

Here's my initial take on how this might work:
- Mount hall effect sensor to car. @adammconway suggested attempting to sense the motor directly, which would imply mounting the sensor in an axial orientation relative to the motor.
- If we can't accurately sense the motor directly, use small magent(s) mounted on the drive shaft or a gear in the drivetrain and mount sensor in an appropriate place given the location of the magnets.
- Connect hall effect sensor signal pin to one of the Pi's GPIO pins that supports interrupt handling
- Connect hall effect power and ground to the corresponding pins on the servo hat

Reading the sensor:
- Configure the interrupt on the Pi with a [callback function](http://raspi.tv/2013/how-to-use-interrupts-with-python-on-the-raspberry-pi-and-rpi-gpio-part-2) (aka ISR - interrupt service routine) that is called when the sensor signal pin changes state. The minimum here would be to read a single state change (RISING or FALLING), additional precision could be achieved by configuring the interrupt to track both rising and falling changes.
- Inside the ISR, update variables for tracking state changes (those that will need to be read by the drive loop will need to be global):
  - (local) Timestamp of previous state change
  - (global) Time elapsed between previous state change and current state change
  - (global) Total count of state changes

Calculating distance and speed:
- Distance traveled will be informed by the relationship between the point in the drive train that is being monitored by the hall effect sensor and the physical distance traveled by the car for 1 cycle (or one half cycle if we monitor both rising and falling edges) of the hall effect sensor signal. We'll need some constant to tie these two together (something like "feet per hall sensor state change"), which will be vehicle specific. Perhaps this is a new addition to the vehicle config file?
- Velocity will be calculated using the distance above divided by the elapsed time that was recorded in the ISR. Calculating speed using the time elapsed from only the most recent sensor cycle would yield the most recent velocity, but we might find that averaging this over several cycles to yield an average velocity over a slightly longer time period would smooth out the data a bit. Let's start with the 1 cycle calculation and then we can adjust if needed.

Sending distance and velocity back to server:
- TBD @wroscoe do you have any thoughts here?

@wroscoe @adammconway does the above approach sound reasonable to you? Do you have preferences on the units used for distance and velocity?

adammconway, 2017-03-02:
Hi @alanwells, this seems to make sense... I will try something similar for the brushless motor, which might be easier because the rotor is actually a magnet.  

As I have thought about how a brushed motor works, it may be hard to get this to work (at a specific location on the can, the coil probably only rotates 60 or 120 degrees, this may not be a big enough change in the b field to trigger the hall effect sensor.  Anyways... still worth experimenting. 

What I found with my ESC tests for the brushless motor controller (not the sensored one) was that there is Hysteresis between acceleration and deceleration, this probably means that you will need to implement a pretty aggressive control loop.  I would suggest a PID loop, just keep in mind that this requires a lot of fiddling with the various Gains to get this to work.  I am happy to help with this/code the PID.

alanwells, 2017-03-03:
I failed to notice that the [hall effect sensor from SparkFun](https://www.sparkfun.com/products/9312) will need a level shifter to work directly with the pi's 3.3V logic level pins. I don't have the level shifter parts on hand, so I won't be able to test this tomorrow at the track day. I will bring the parts I ordered, if any of you have a level shifter ([something like this](https://www.sparkfun.com/products/12009)), we might be able to get it hooked up.

alanwells, 2017-03-03:
@adammconway Thanks for the advice here. I've never implemented a PID loop, your help would be much appreciated.

alanwells, 2017-05-08:
@adammconway @wroscoe I have my hall effect sensor mounted and the circuit is working w/ state changes triggered by magnets mounted on the drive shaft. Now it's on to the software side. I think there are at least four pieces to this:

1) Additions to vehicle.ini that enables odometry and defines the wheel circumference to use for distance calculations.
2) An interrupt that counts the wheel rotations and timing data using the output from the hall effect sensor and stores the data in a (global?) variable that can be accessed by the main drive loop
3) A method that calculates velocity based on the last n number of wheel revolutions (as tracked by the interrupt), the timestamps of those state changes, and the circumference of the wheel provided in vehicle.ini.
4) A PID controller class. I think this needs to take current PWM, current velocity, and target velocity as inputs, and should output the new PWM value.
5) Possibly some feedback mechanism to pass velocity (and distance traveled?) back to the server for display in the controller UI, and to be used for labeling images (so we can attempt to train the model to infer velocity w/o hall effect sensor).

Questions for you guys:
1) Does the above description sound reasonable? Are there any modifications that you would suggest?
2) Should we calculate velocity based on only the last rotation or average it over a number of rotations (possibly less precise but likely also less noisy?)
3) From an architecture perspective, it seems like the PID controller should be it's own class, which will then be an optional input into the initialization of vehicle object created [here](https://github.com/wroscoe/donkey/blob/master/scripts/drive.py#L62) in `drive.py` (optional because not every car will have odometry). Does this sound right?

adammconway, 2017-05-08:
I think a float that is something like rising and falling edges per meter or something like that would make sense… I can imagine designs that will have multiple pulses per revolution and also there is a gear ratio between input into the differential and output.

wroscoe, 2017-08-14:
I believe this was implemented with a rotory encoder and is included as a part.

closed 2017-08-14 by wroscoe