autorope/donkeycar #73: Allow multiple vehicle profiles via multiple .ini files

yconst, 2017-03-13:
The vehicle could be passed as an optional parameter when running the script files, otherwise the default (vehicle.ini) would be used.

wroscoe, 2017-03-13:
When would you need multiple profiles on a single vehicle? Also, you should join our slack channel to discuss and share  your builds. Enter your email here and I'll add you: http://donkeycar.weebly.com/

yconst, 2017-03-14:
Thanks, I've sent a request.

The case I was facing was that of having two vehicles and changing the SD card between them. One had different servo/motor settings so I believe multiple configs can be helpful in such cases. 

But I imagine this could also go another way: Offering different stock configuration files for different vehicle models (e.g. different rc car models) and types (diff steering, normal steering), can open Donkey to a wider audience (e.g. people who already have different models of rc cars/vehicles).

wroscoe, 2017-03-14:
This should be done with an optional argument "--config" that can specify a different config file. The diff drive does not yet use config files.

yconst, 2017-03-21:
I've added --config parsing and a couple of vehicle configs based on the google doc manual in the following commit:
https://github.com/yconst/donkey/commit/5fb192c443eb0b43ea6d37134cb0a800bc263ff8

I did not create a pull request because this commit also includes the changes to start-server.sh, part of another mr. Once that mr is resolved, I'm going to add one for this too.

yconst, 2017-04-12:
This has been merged to dev. Closing

closed 2017-04-12 by yconst