autorope/donkeycar #76: Error starting drive.py

rlyons20, 2017-03-22:
I've tested my motors, using the Adafruit example scripts, and everything is working.

When I get to the last step of starting the remote control, this is the response:

`Traceback (most recent call last):
  File "scripts/drive.py", line 32, in <module>
    mythrottlecontroller = dk.actuators.PCA9685_Controller(cfg['throttle_actuator_channel'])
  File "/home/bob/donkey/donkey/actuators.py", line 34, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685()
  File "/home/bob/donkey/env/lib/python3.4/site-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/home/bob/donkey/env/lib/python3.4/site-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/home/bob/donkey/env/lib/python3.4/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/bob/donkey/env/lib/python3.4/site-packages/Adafruit_PureIO/smbus.py", line 236, in write_byte_data
    self._device.write(data)
OSError: [Errno 5] Input/output error`

wroscoe, 2017-03-22:
I haven't seen this error before but others (outside this project) have had that error:
* http://stackoverflow.com/questions/30325351/ioerror-errno-5-input-output-error-while-using-smbus-for-analog-reading-thr
* https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=17295

I suspect you have a process already using the I2C that conflicts with the attempted write. Try restarting "sudo restart' the pi.

rlyons20, 2017-03-22:
I've restarted a bunch :(

I am using this stepper hat (https://www.adafruit.com/products/2348), instead of the one you spec in your build. Think that might be a problem?

wroscoe, 2017-03-22:
Yes. Thats the motor hat for a differential drive car (one that uses DC motors).

closed 2017-03-23 by wroscoe