autorope/donkeycar #82: Docker instance does not update git repo when started.

wroscoe, 2017-04-25:
People who use the docker image to start the server don't use an updated version of the git repo. The repo can be updated manually by running 
``` 
bash start-server.sh -d
```
and then running the following inside the docker instance.
```
git pull origin master
```
but it would be better if this updated automatically, assuming we can keep the master branch free of conflicts.  @yconst  do you know how to do this.

yconst, 2017-04-26:
I think if one does a `git pull` and then normally start the server, i.e. `./start-server.sh` , it should update the tree inside the image automatically (the only drawback being that it runs `pip install` again). I tend to believe that this should be the standard way of doing things (i.e. the docker tree always mirrors the local tree), otherwise there is the risk of conflicts as you mentioned.

yconst, 2017-04-26:
Actually the way the Dockerfile is configured it should only trigger a `pip install` if setup.py changes.

yconst, 2017-04-26:
Or we could completely switch to using the `-v` flag in the startup script (and lose a bit of portability along the way)

binary-bard, 2017-06-15:
Are there any downsides to mapping the directory (just like mydonkey) to docker image? That way, one can do git pull without needing to refresh/rebuild docker image.

tizianofiorenzani, 2017-07-08:
Guys, for people like me that have never used docker before, would be great to have it mentioned at least  in the readme

wroscoe, 2017-07-09:
Awesome feedback. Can you make a pull request?

-Will

wroscoe, 2017-08-14:
The new version doesn't require docker as everything is run on the pi.

closed 2017-08-14 by wroscoe