autorope/donkeycar #88: Create VehicleState class to generalize the data record/retrieve functions.

wroscoe, 2017-06-04:
At the next race (june 16th), several of us plan to implement lidar + odometry on Donkey2 cars. However the current donkey software doesn't support adding additional inputs since the record/decided functions are hardcoded (angle, steering, img_arr). A generalized state class would allow the cars to support data from additional sensors and allow flexible data retrieval (ie images from last 10 frames).

This VehicleState class would act like a ROSBag and could save data like this:
``` python
state = VehicleState(save_to='/path/to/session/')
state.put('image', img_arr)
state.put('throttle', throttle)
state.put('angle', angle)
```

VehicleState saves data to disk depending on the type of data. 1 or 3 channel arrays (images) would be saved as jpg images. Single value data (speed, throttle would be saved to a csv file that contains a list of all values in the format: 
```
key, value, time
throttle, .23, 12:32:32
angle, -.1, 12:32:33
image, /path/to/image, 12:32:33
throttle, .23, 12:32:35
angle, -.1, 12:32:36
image, /path/to/image, 12:32:37
``` 

You could then retrieve the last variable values like this:
``` python
img_arr, throttle, angle = state.get(['image', 'throttle', 'angle'])
```

Since the variables are not being recorded at the same time the state class would need to interpolate the different data sets to create a tabular output that Keras / Tensorflow needs.

mw46d, 2017-06-04:
I'm not sure about the `time' value. Probably better to save  milliseconds since start of the run or something like that

mw46d, 2017-06-04:
Overall, I would probably prefer a bit more structure. Maybe all values at the time of a given image.  Maybe a JSON or similar, something that translates easily into a Python dict where you can look at what you want and ignore what you don't expect