autorope/donkeycar #94: mydonkey/models folder missing

tizianofiorenzani, 2017-07-09:
Hi everyone, I just cloned on Ubuntu 14, installed and run docker. When I bash start-server.sh I get:

.........
start-server: Running Donkey server container...
Loading modules for server.
Starting Donkey Server...
Using TensorFlow backend.
Traceback (most recent call last):
  File "/donkey/scripts/serve.py", line 12, in <module>
    w = dk.remotes.DonkeyPilotApplication()
  File "/donkey/donkey/remotes.py", line 175, in __init__
    self.pilots = ph.default_pilots()
  File "/donkey/donkey/pilots.py", line 84, in default_pilots
    pilot_list = self.pilots_from_models()
  File "/donkey/donkey/pilots.py", line 71, in pilots_from_models
    models_list = [f for f in os.scandir(self.models_path)]
FileNotFoundError: [Errno 2] No such file or directory: '/root/mydonkey/models'

wroscoe, 2017-07-09:
There is a work around described here: https://stackoverflow.com/questions/44557347/why-do-i-get-an-error-when-i-try-to-run-donkeycar-server-using-the-docker-image

wroscoe, 2017-08-14:
This is no longer an issue in the new version that doesn't require docker.

closed 2017-08-14 by wroscoe