autorope/donkeycar #107: Feature request - Servo control via PPM instead of I2C

dlarue, 2017-08-07:
There have been some comments about swapping in/out the stock RC controller and since there are only 2 servo controls used( steering / throttle ) an Arduino could do this.  But since the rPi has only 1 hardware PWM, it would be more efficient to stream the servo control via a single PPM stream. This way an Arduino could decode the PPM and drive the servo's.  And with a single digial I/O flag/pin the Arduino could read and use the stock RC receiver signals for training.

zlite, 2017-08-07:
That's what we do with OpenMV, and the Arduino allows both multichannel PWM
output and a wide range of inputs (analog, digital, serial, etc).  We
implement Firmata <https://www.arduino.cc/en/Reference/Firmata> on the
Arduino side, and recommend that be used as the the middleware standard for
I/O.

A standard Arduino Pi ha
<http://www.robotshop.com/en/arduino-adapter-raspberry-pi.html?gclid=Cj0KCQjwn6DMBRC0ARIsAHZtCeOvuKO5HPcGp-gxiCQi4-hXyDFFxUjqCWcjBu6Y631-Ygxw_Lki9d0aAnN7EALw_wcB>t
can work, although a custom shield like we designed for the OpenMV
<https://circuits.io/circuits/4446963-openmv-arduino-shield> simplifies the
wiring.

dlarue, 2017-08-07:
The more I thought about it, I think keeping I2C from the rPi makes sense since all I'm looking to do is add simple RC control back and if I can do it with a $3 ProMini and replace the Servo Driver($12) too then it's a feature improvement and cost cutter.  The RC control could also enable a remote kill switch via a 3rd channel.

As for adding sensors, an Arduino with Firmata sounds interesting. I'm only familiar with Firmata by using it with Snap4Arduino(Scratch for Arduino).  And idea to keep in the back pocket. Thanks.

closed 2017-08-21 by wroscoe