autorope/donkeycar #110: Update web controller to use sockets.

wroscoe, 2017-08-17:
It would be helpful to have the car send messages to the web app used to drive the car. Advantages include. 
* Car could know if connection was lost.
* Vehicle status could be sent to web app. 

The changes could all be confined to the [LocalWebController](https://github.com/wroscoe/donkey/blob/master/donkeycar/parts/controllers/web.py#L102).

igy254, 2018-08-13:
The correct URL to the file where you could confine all changes is [LocalWebController](https://github.com/wroscoe/donkey/blob/master/donkeycar/parts/web_controller/web.py). By the way @wroscoe I could help you with this one if no one takes it. I guess to test this one I need a car, and my should be shipped soon.

wroscoe, 2018-11-22:
This code is moved here https://github.com/autorope/donkeypart_web_controller

closed 2018-11-22 by wroscoe