autorope/donkeycar #111: when setting MAX throttle, it appears the joystick max throw is not set to the new MAX value

dlarue, 2017-08-17:
I've slowed down the bot to 25% and driving from the phone or laptop web page it feels like it hits full 25% throttle in the first 25% movement of the joystick. Rescaling the MAX throttle value to the max throttle position would make driving much easier.

wroscoe, 2017-08-18:
@dlarue You are correct. The current max throttle just acts as a governor. It's also used to set a constant throttle when using that throttle mode. This is so we can race in DIYRobocars events since most the current models don't control the speed and the competition requires all hands off the controls. 

A setting on the web control that would tune the sensitivity of the throttle would be helpful. Currently you can do this by lowering your max_pulse of your throttle actuator. https://github.com/wroscoe/donkey/blob/master/donkeycar/templates/donkey2.py#L80

dlarue, 2017-08-18:
It makes sense to have a way to fix a steady throttle for when in autonomous mode but the throttle for training is tied to the GUI and shouldn't that operate between 0 and what ever is the operational full speed?

I've ordered a PS3 type Bluetooth controller and will check with the author off the code used to drive that to see if it works better for training.

Ezward, 2023-02-18:
Looks like this was done by @tawnkramer on Dec 14 2017.  The throttle limit is now applied as scale to the throttle value provided by the joystick.  https://github.com/autorope/donkeycar/blob/c4d90910e0e1b5056dc8961a09c7e5eb1ba18ffe/donkeycar/parts/controller.py#L1024 Thanks Tawn.

closed 2023-02-18 by Ezward