autorope/donkeycar #112: VNH5019 and direct control via GPIO

wallarug, 2017-08-21:
Hey,

I'm working on using a Polulu Dual VNH5019 Motor Driver Shield with the donkey car code.  How easy do you think it will be to integrate it to the actuators.py and the project as a whole?

I am in the process of writing a driver similar to the adafruit PCA9685 python library for the VNH5019.  However, the one key difference is that the VNH5019 does not have I2C... it would require using the Raspberry Pi GPIO directly.  From what I can see here, this has not been done before for this project.  

I haven't had enough time to comb through all the code on here, so please point me in the right direction if this is functionality already implemented.

1. Can I implement my driver to use the GPIO pins directly (not I2C) with the existing code?
2. How / if the driving component would work under this method?

Thanks in advance.

PS:  Great project and body of code!

wroscoe, 2017-08-21:
This sounds doable and there are examples of accessing the GPIO channels from python. https://sourceforge.net/p/raspberry-gpio-python/wiki/BasicUsage/

To implement you'd create a new actuator part class like those listed here: https://github.com/wroscoe/donkey/blob/master/donkeycar/parts/actuators/actuators.py

Then just use that part in your car script instead of the current actuator. https://github.com/wroscoe/donkey/blob/master/donkeycar/templates/donkey2.py#L75

What are the advantages of this proposed change?

wallarug, 2017-08-22:
Hey @wroscoe ,

Thanks for getting back to me.

The only advantage of this proposed change is improved portability between different hardware for the project as a whole.   The VNH5019 is a very powerful motor driver (12 Amps) which could be helpful for people who want to make faster cars.  This could also open up the possibility for documentation for others to write drivers for other chips to work with the donkey car project. 

I'm happy to implement this enhancement, since you have highlighted the relevant sections that can be changed above.  I will let you know once I am done and you can decide to include the enhancement or not.

wallarug, 2017-09-29:
Hey @wroscoe,

I've made significant progress on my driver for this device.  I'm just testing it at the moment to make sure that it works as expected.  I'll update you when it is working.

arunderwood, 2017-10-23:
@wallarug I'm currently working on wiring up a dual VNH2SP30 board to a PowerWheels car, so your progress on abstracting hardware support is very interesting to me.  I'm curious, what are you hooking the VNH5019 up to?

wallarug, 2017-10-23:
Hey @arunderwood ,

I have it all working and hocked up to the Raspberry Pi's GPIO pins.  I think my motor shield is playing up a bit.  If you could test it that would be great.

I've dumped all the code onto the repository here:  https://github.com/wallarug/donkey-vnh5019

Feel free to contact me if you need any help.

arunderwood, 2017-10-23:
@wallarug awesome!  I'm new to the Donkey codebase so it will take me a little bit to wrap my head around everything.

What are your thoughts on forking wroscoe/donkey and making the hardware abstraction changes on the fork?  That way it's easier for me/others checkout the code and run it all in one go, and those changes can be PR'ed back to mainline if we can get it to a state Will is happy with.

wallarug, 2017-10-28:
Hey @arunderwood ,

I'm happy to do that, but it will have to wait. I'll create a pull request and add all my code when I get some free time.

tawnkramer, 2019-06-12:
closing due to inactivity. please open if there is more to discuss.

closed 2019-06-12 by tawnkramer