autorope/donkeycar #116: building on rPi fails, can't find hdf5.h

dlarue, 2017-08-28:
after clone called donkeycar and than adding sudo apt-get install python-dev.

I run sudo pip install -i ./donkeycar

but it fails with this:
In file included from /tmp/pip-build-oqC9vR/h5py/h5py/defs.c:515:0:

/tmp/pip-build-oqC9vR/h5py/h5py/api_compat.h:27:18: fatal error: hdf5.h: No such file or directory

 #include "hdf5.h"

                  ^

compilation terminated.

dlarue, 2017-08-28:
trying it again after installing libhdf5-dev via apt.

dlarue, 2017-08-28:
that appears to have allowed the build to finish. Only now, when I run donkey createcar --path ~/d2b it fails with a scipy error:

(env)pi@dougs-robocar:~/donkey $ donkey createcar --path ~/d2b
Traceback (most recent call last):
  File "/usr/local/bin/donkey", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2876, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 449, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 745, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 639, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: scipy>=0.14

And I tried using "sudo pip3 install scipy -U" to get v 0.19 but that fails with the dreaded, "Storing debug log for failure in /root/.pip/pip.log" failure:

Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-svVy85/scipy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-rFgVnR-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-svVy85/scipy
Storing debug log for failure in /root/.pip/pip.log

dlarue, 2017-08-28:
sudo apt-get install libatlas-base-dev gfortran
sudo pip3 install scipy -U
...

tawnkramer, 2017-09-18:
if you want to do a manual install, I recommend anaconda. http://docs.donkeycar.com/faq/ has an approach that worked for me.

closed 2017-09-18 by tawnkramer