autorope/donkeycar #175: train fails to pick up all tubs

tawnkramer, 2017-11-06:
when no tub arg given, train should use all tub in default data dir.

(donkey) C:\cygwin\home\tkramer\projects\d2.2>python manage.py train --model models/test.h5
using donkey v2.2.0 ...
Using TensorFlow backend.
loading config file: C:\cygwin\home\tkramer\projects\d2.2\config.py
config loaded
path_list: None
Traceback (most recent call last):
  File "manage.py", line 183, in <module>
    train(cfg, tub, model)
  File "manage.py", line 148, in train
    tubgroup = TubGroup(tub_names)
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 644, in __init__
    tub_paths = self.resolve_tub_paths(tub_paths)
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 677, in resolve_tub_paths
    path_list = path_list.split(",")
AttributeError: 'NoneType' object has no attribute 'split'

wroscoe, 2017-11-06:
Yep, I removed this because I think explicit behavior is better. To train on all the tubs you need to edit the command to `python manage.py train --tubs '~/d2/tubs/*' --model models/test.h5`

wroscoe, 2017-11-07:
I added a fix for this in dev.

closed 2017-11-07 by wroscoe