autorope/donkeycar #176: train fails to load tub specified

tawnkramer, 2017-11-06:
when given a single tub dir, fails to open tub

(donkey) C:\cygwin\home\tkramer\projects\d2.2>python manage.py train --tub=data\tub_1_17-11-03 --model=models\test.h5
using donkey v2.2.0 ...
Using TensorFlow backend.
loading config file: C:\cygwin\home\tkramer\projects\d2.2\config.py
config loaded
path_list: data\tub_1_17-11-03
Traceback (most recent call last):
  File "manage.py", line 183, in <module>
    train(cfg, tub, model)
  File "manage.py", line 148, in train
    tubgroup = TubGroup(tub_names)
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in __init__
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in <listcomp>
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 179, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check
ide meta info to create a new tub.

tawnkramer, 2017-11-06:
path mask also fails:

(donkey) C:\cygwin\home\tkramer\projects\d2.2>python manage.py train --tub=data\* --model=models\test.h5
using donkey v2.2.0 ...
Using TensorFlow backend.
loading config file: C:\cygwin\home\tkramer\projects\d2.2\config.py
config loaded
path_list: data\*
Traceback (most recent call last):
  File "manage.py", line 183, in <module>
    train(cfg, tub, model)
  File "manage.py", line 148, in train
    tubgroup = TubGroup(tub_names)
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in __init__
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in <listcomp>
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 179, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check your tub path or prov
ide meta info to create a new tub.

tawnkramer, 2017-11-06:
comma separated tubs also fail:
```
(donkey) C:\cygwin\home\tkramer\projects\d2.2>python manage.py train --tub=data\tub_1_17-11-03,data\tub_2_17-11-03 --model=models\test.h5
using donkey v2.2.0 ...
Using TensorFlow backend.
loading config file: C:\cygwin\home\tkramer\projects\d2.2\config.py
config loaded
path_list: data\tub_1_17-11-03,data\tub_2_17-11-03
Traceback (most recent call last):
  File "manage.py", line 183, in <module>
    train(cfg, tub, model)
  File "manage.py", line 148, in train
    tubgroup = TubGroup(tub_names)
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in __init__
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 645, in <listcomp>
    tubs = [Tub(path) for path in tub_paths]
  File "c:\cygwin\home\tkramer\projects\donkey\donkeycar\parts\datastore.py", line 179, in __init__
    raise AttributeError(msg)
AttributeError: The tub path you provided doesn't exist and you didnt pass any meta info (inputs & types)to create a new tub. Please check your tub path or prov
ide meta info to create a new tub.
```

wroscoe, 2017-11-06:
I'll add tests for these to see if I can replicate. Masking without quotes wasn't working for me but with quotes it worked.

tawnkramer, 2017-11-07:
All three of these cases used to work.

wroscoe, 2017-11-07:
@tawnkramer The comma separated and single tub work for me. Can you check that your paths were correct? I can't reproduce those. For masking it's OK to require quotes. I've added this to the help command. Closing, reopen if you can reproduce.

tawnkramer, 2017-11-07:
If I add a line @ 153 in datastore.py
```
    def __init__(self, path, inputs=None, types=None):
        print('tub is using path:', path)
        self.path = os.path.expanduser(path)
        self.meta_path = os.path.join(self.path, 'meta.json')
        self.df = None
```
then I can see it's somehow duplicated the path on itself:
```
(env) tkramer@tkramer-VirtualBox:~/d2.2$ python manage.py train --tub ./data/left_lane_hive_track_09_02_2017_tkramer --model models/test.h5
using donkey v2.2.0 ...
Using TensorFlow backend.
loading config file: /home/tkramer/d2.2/config.py
config loaded
tub_names ./data/left_lane_hive_track_09_02_2017_tkramer
path_list: ./data/left_lane_hive_track_09_02_2017_tkramer
tub is using path: ./data/left_lane_hive_track_09_02_2017_tkramer/./data/left_lane_hive_track_09_02_2017_tkramer
```

frankinulm, 2017-11-12:
have the same problem - it seems to be within def find_tub_paths(self, path)
`    def find_tub_paths(self, path):
        matches = []
        path = os.path.expanduser(path)
        for file in glob.glob(os.path.join(path)):
            print('Loop:',file)
            if os.path.isdir(file):
                matches.append(os.path.join(path, file))
            print('Loop: Matches',matches)

        return matches`
The append joins the path (which already contains the filename) with the filename
Unfortunately my Python is quite poor - don't want to change it fearing some side effects

wroscoe, 2017-11-12:
This is fixed in the dev branch.

wroscoe, 2017-11-13:
V 2.2.1 is now merged to master.

closed 2017-11-13 by wroscoe