autorope/donkeycar #193: part: cv.py ImgCrop class reduces width regardless of inputs

wheatgrinder, 2017-12-06:
donkey/donkeycar/parts/cv.py

in the code below the width and height values are in the wrong order, the img_array.shape call returns height first.. so the width of the output img_arry is always set to the input images height, typically 120

```
class ImgCrop:
    """
    Crop an image to an area of interest. 
    """
    def __init__(self, top=0, bottom=0, left=0, right=0):
        self.top = top
        self.bottom = bottom
        self.left = left
        self.right = right
        
    def run(self, img_arr):
        width, height, _ = img_arr.shape  # width and height are reversed, resulting in the issue
        img_arr = img_arr[self.top:height-self.bottom, 
                          self.left: width-self.right]
        return img_arr
```

wroscoe, 2017-12-06:
Since you know how to fix this, could you submit a pull request? Thanks.

wheatgrinder, 2017-12-07:
hey, i think I got it right.. 
Forked the code
fixed the issue
pushed to my forked repo
create pull request.. is that how its done?

wroscoe, 2018-07-31:
I'm removing opencv from the repo. This can be implemented with a keras layer or tensor operation.

closed 2018-07-31 by wroscoe