autorope/donkeycar #200: Problem in training the autopilot

mxheikki, 2017-12-31:
Hi @wroscoe ,
When I train the autopilot in my local computer, I get the error message (under). Is there anything that I could do to fix the problem? I use Mac Sierra as host computer (donkey 2.2.1), software should  be up to date. Thank you for your awesome project, I am physics teacher, and I am trying to do this with high school students in Finland. Yours, mxheikki

  File "/User../projects/donkey/donkeycar/parts/keras.py", line 67, in train
    validation_steps=steps*(1.0 - train_split))
  File "/Users/../miniconda3/envs/donkey/lib/python3.6/site-packages/keras/legacy/interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "/Users/../miniconda3/envs/donkey/lib/python3.6/site-packages/keras/engine/training.py", line 1840, in fit_generator
    class_weight=class_weight)
  File "/Users/..../miniconda3/envs/donkey/lib/python3.6/site-packages/keras/engine/training.py", line 1559, in train_on_batch
    check_batch_axis=True)
  File "/Users/.../miniconda3/envs/donkey/lib/python3.6/site-packages/keras/engine/training.py", line 1234, in _standardize_user_data
    exception_prefix='input')
  File "/Users/.../miniconda3/envs/donkey/lib/python3.6/site-packages/keras/engine/training.py", line 140, in _standardize_input_data
    str(array.shape))
ValueError: Error when checking input: expected img_in to have shape (None, 120, 160, 3) but got array with shape (128, 160, 120, 3)

rlehman221, 2017-12-31:
Were you able to find a solution to this?

wroscoe, 2017-12-31:
Early december I changed the camera config to input height and with for consistency through the project. It looks like you recorded images with the old settings and so the images don't match the model inputs. You can fix this by recording images after changing camera settings on your Pi  in the config.py  to be `CAMERA_RESOLUTION = (120, 160)`

closed 2017-12-31 by wroscoe