autorope/donkeycar #206: ImportError: No Module named docopt

pj2br, 2018-01-16:
Every time I would like to run the python manage.py train --tub "data/tub_1*" --model ./models/mypilot script, it gives me:
ImportError: No module named docopt
Any ideas/tips?

wroscoe, 2018-01-17:
I suspect your virtualenv is not activated or donkeycar was not installed in the virtualenv. "docopt" is listed as a requirement in the donkeycar setup.py file. This command should be run on your computer not your pi

closed 2018-01-17 by wroscoe

pj2br, 2018-01-17:
Hi Will,

Thank you for answering me! I am really a novice here; would you be able to guide me through this issue? I followed the documents to the very last letter (did it on a Win7, Mac and Linux machines). How do I activate this virtualenv? On the Win7 machine, I was able to start the manage.py train script. After reaching Epoch 22/100 it will issue an ‘early stop’ message. Tried it several times and it will not pass Epoch 27/100.
Really appreciate your time and patience in helping me out- this project is part of my school graduation project.

Regards,

Brett

Closed #206<https://github.com/wroscoe/donkey/issues/206>.

wroscoe, 2018-01-17:
Early stop means that the validation error stopped improving and you're
model should be ready to go. Did you try it?

Seperatly, if you get a No Module error you can do "pip install <name of
module>" to install the module and usually fix that error. You can read up
on your own about virtualenvs. There are lots of good resources on it.

Also when you're looking for help. Post the terminal commands that you've
tried and their output so people can see whats going on.

Good luck!

- Will

pj2br, 2018-01-17:
Dear Will,

Thanks for your help!! We gave it another shot this afternoon. Did not load it up to the Pi/car as yet. After about 2 hours of crunching, it stopped , see photo. If you should see any ‘strange’ things in this photo, please let me know.
[image1.jpeg]
I will share my results as soon as I am done uploading and testing the car.

Thanks again!!

Brett