autorope/donkeycar #211: Freeze when start drive

mrtortong, 2018-01-27:
Hi
I found Freeze when start drive
(dk) pi@d2:~/d2 $ python manage.py drive --js
using donkey v2.2.1 ...
Using TensorFlow backend.
loading config file: /home/pi/d2/config.py
config loaded
PiCamera loaded.. .warming camera
Adding part PiCamera.
Adding part JoystickController.
Adding part Lambda.
Adding part KerasCategorical.
Adding part Lambda.
Adding part PWMSteering.
Adding part PWMThrottle.
path_in_tub: /home/pi/d2/data/tub_5_18-01-27
Tub does NOT exist. Creating new tub...
New tub created at: /home/pi/d2/data/tub_5_18-01-27
Adding part TubWriter.
Opening /dev/input/js0...
Starting vehicle...
Device name: Logitech Gamepad F710
/home/pi/.virtualenvs/dk/lib/python3.4/site-packages/picamera/encoders.py:544: PiCameraResolutionRounded: frame size rounded up from 120x160 to 128x160
  width, height, fwidth, fheight)))

wroscoe, 2018-01-28:
There is not enough here to identify and fix this bug.

closed 2018-01-28 by wroscoe

scitronboy, 2018-01-29:
apichat, I think this is normal. go to <your pi's ip address>:8887, and it
should work. my log always says that and it works.