autorope/donkeycar #215: support for Asus Tinkerboard?

supsup, 2018-02-07:
I went over the FAQs, and installed bazel/tensorflow from source.
I had to change a few of the steps but it all installed well.
I generated the wheel file and installed.

I'm at this step but it complains about my system not being a pi:
`
pip install -e .[pi] 
 `
is there a command for the tinkerboard like this? 
Can i just skip to the next part?

supsup, 2018-02-07:
here is the output I am getting:

`Obtaining file:///home/tinker1/donkey/donkey
Requirement already satisfied: numpy in /usr/local/lib/python3.5/dist-packages (from donkeycar==2.2.1)
Collecting pillow (from donkeycar==2.2.1)
  Using cached Pillow-5.0.0.tar.gz
Collecting docopt (from donkeycar==2.2.1)
Collecting tornado (from donkeycar==2.2.1)
Collecting requests (from donkeycar==2.2.1)
  Using cached requests-2.18.4-py2.py3-none-any.whl
Requirement already satisfied: keras in /home/tinker1/donkey/env/lib/python3.5/site-packages (from donkeycar==2.2.1)
Requirement already satisfied: h5py in /usr/lib/python3/dist-packages (from donkeycar==2.2.1)
Collecting python-socketio (from donkeycar==2.2.1)
  Using cached python_socketio-1.8.4-py2.py3-none-any.whl
Collecting flask (from donkeycar==2.2.1)
  Using cached Flask-0.12.2-py2.py3-none-any.whl
Collecting eventlet (from donkeycar==2.2.1)
  Using cached eventlet-0.22.0-py2.py3-none-any.whl
Collecting moviepy (from donkeycar==2.2.1)
  Using cached moviepy-0.2.3.2-py2.py3-none-any.whl
Requirement already satisfied: pandas in /usr/lib/python3/dist-packages (from donkeycar==2.2.1)
Collecting picamera (from donkeycar==2.2.1)
  Using cached picamera-1.13.tar.gz
Collecting Adafruit_PCA9685 (from donkeycar==2.2.1)
Collecting certifi>=2017.4.17 (from requests->donkeycar==2.2.1)
  Using cached certifi-2018.1.18-py2.py3-none-any.whl
Collecting urllib3<1.23,>=1.21.1 (from requests->donkeycar==2.2.1)
  Using cached urllib3-1.22-py2.py3-none-any.whl
Collecting idna<2.7,>=2.5 (from requests->donkeycar==2.2.1)
  Using cached idna-2.6-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests->donkeycar==2.2.1)
  Using cached chardet-3.0.4-py2.py3-none-any.whl
Requirement already satisfied: pyyaml in /home/tinker1/donkey/env/lib/python3.5/site-packages (from keras->donkeycar==2.2.1)
Requirement already satisfied: scipy>=0.14 in /usr/lib/python3/dist-packages (from keras->donkeycar==2.2.1)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3/dist-packages (from keras->donkeycar==2.2.1)
Collecting python-engineio>=1.2.1 (from python-socketio->donkeycar==2.2.1)
  Using cached python_engineio-2.0.2-py2.py3-none-any.whl
Requirement already satisfied: Werkzeug>=0.7 in /usr/local/lib/python3.5/dist-packages (from flask->donkeycar==2.2.1)
Collecting click>=2.0 (from flask->donkeycar==2.2.1)
  Using cached click-6.7-py2.py3-none-any.whl
Collecting itsdangerous>=0.21 (from flask->donkeycar==2.2.1)
Collecting Jinja2>=2.4 (from flask->donkeycar==2.2.1)
  Using cached Jinja2-2.10-py2.py3-none-any.whl
Collecting greenlet>=0.3 (from eventlet->donkeycar==2.2.1)
Collecting enum34 (from eventlet->donkeycar==2.2.1)
  Using cached enum34-1.1.6-py3-none-any.whl
Collecting decorator==4.0.11 (from moviepy->donkeycar==2.2.1)
  Using cached decorator-4.0.11-py2.py3-none-any.whl
Collecting imageio==2.1.2 (from moviepy->donkeycar==2.2.1)
Collecting tqdm==4.11.2 (from moviepy->donkeycar==2.2.1)
  Using cached tqdm-4.11.2-py2.py3-none-any.whl
Collecting Adafruit-GPIO>=0.6.5 (from Adafruit_PCA9685->donkeycar==2.2.1)
Collecting MarkupSafe>=0.23 (from Jinja2>=2.4->flask->donkeycar==2.2.1)
Collecting spidev (from Adafruit-GPIO>=0.6.5->Adafruit_PCA9685->donkeycar==2.2.1)
Collecting adafruit-pureio (from Adafruit-GPIO>=0.6.5->Adafruit_PCA9685->donkeycar==2.2.1)
Building wheels for collected packages: pillow, picamera
  Running setup.py bdist_wheel for pillow: started
  Running setup.py bdist_wheel for pillow: finished with status 'error'
  Complete output from command /home/tinker1/donkey/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8ezrrps0/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp8z53a_1gpip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/TarIO.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-3.5/PIL
  copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-3.5/PIL
[... 381 lines cut ...]
        cmd_obj.run()
      File "/usr/lib/python3.5/distutils/command/build_ext.py", line 338, in run
        self.build_extensions()
      File "/tmp/pip-build-8ezrrps0/pillow/setup.py", line 580, in build_extensions
        raise RequiredDependencyException(f)
    __main__.RequiredDependencyException: jpeg
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-8ezrrps0/pillow/setup.py", line 804, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:
    
    The headers or library files could not be found for jpeg,
    a required dependency when compiling Pillow from source.
    
    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html
    
    
    
    ----------------------------------------
`

scitronboy, 2018-02-08:
> File "", line 1, in

how are you even getting the picamera to work with the tinkerboard?

tikurahul, 2018-07-20:
@supsup Did you make more progress on installing `donkeycar` on the `Tinkerboard` ?

tikurahul, 2018-07-23:
Nevermind. I figured it out. `donkeycar` now works with the Tinkerboard S. I had to tweak a few dependencies.

wroscoe, 2018-07-31:
There are already too many frays in the donkey's mane. The project default is not going to support another board right now. Don't let this stop you from trying though.

closed 2018-07-31 by wroscoe

tikurahul, 2018-07-31:
The Tinkerboard S just works for me. I just followed instructions from [here](https://github.com/wroscoe/donkey/blob/dev/install/pi/install.sh). Make sure you are not running Python 3.7, as Tensorflow does not support that version of Python yet. 

Using the `PiCamera` may be slightly difficult, but I am using a `JeVois` camera (and I authored a donkey part for that). Hopefully this will get you going.