autorope/donkeycar #220: Required parts: Usb battery link is broken

tmandekic, 2018-03-08:
Link for the usb battery is broken and I'm (a newbie in hardware assembly and) not sure what can I substitute it with. 
Not working: https://www.amazon.com/gp/product/B00P7N0320?tag=donkeycar-20
Could please provide a link to usb battery that can be used for assembly? 
Thanks in advance!

wroscoe, 2018-03-08:
Thanks. I'll fix this soon.

closed 2018-03-08 by wroscoe

Bleyddyn, 2018-03-12:
Not sure where else to ask this. Do you know if that battery can be charged at the same time it's being used to power the Raspberry Pi? A lot of USB phone charging batteries don't work that way and the Amazon pages rarely if ever mention it. It's not a deal breaker but I find it's incredibly useful to be able to charge the battery while still working on the robot.

wroscoe, 2018-03-12:
This battery can not be charged at the same time as you use it. :\

scitronboy, 2018-03-15:
Bleyddyn, you shouldn't even need to do that. if you have it connected with
a USB cord in the first place, why don't you just use one USB cord to
charge the battery and a second USB cord for the raspberry pi??

Bleyddyn, 2018-03-15:
@scitronboy I may not be understanding what you're suggesting. I have mains->battery and battery->rPi at the same time. With my old-no-longer-available PowerGen battery that works. With lots of other batteries it doesn't. As soon as you plug mains power into the battery to start charging it, the battery shuts off its power to the rPi.

So in order to use other batteries I have to: shut down the rPi, rearrange USB cables so they are both plugged into mains and reboot the rPi, then when I want to switch back to battery power I need to shut down the rPi again, rearrange cables and reboot.

commit 3a54b707 (2018-03-08): update battery link
fixes #220