autorope/donkeycar #228: error: unable to create file Icon?: Invalid argument

dschwee, 2018-03-23:
On a windows machine, inside Anaconda Prompt, when I issue the command: 
git clone https://github.com/wroscoe/donkey
The clone succeeds but the checkout fails with a series of errors starting with:
error: unable to create file Icon?: Invalid argument
error: unable to create file docs/Icon?: Invalid argument
error: unable to create file docs/assets/Icon?: Invalid argument
error: unable to create file docs/assets/build_hardware/Icon?: Invalid argument
etc.  
It appears that the package contains filenames that contains question marks which Windows doesn't allow.  
Thoughts?

wroscoe, 2018-03-23:
These empty icon files look like they were added by mistake and not needed. I've deleted them. Can you try cloning again and close this issue if it works. Thanks.

dschwee, 2018-03-23:
Wow!  Thanks for the quick response.  I am away from my donkey system now,
but will clone again this evening and close this issue if it works.
Thanks again!

dschwee, 2018-03-24:
I tried cloning again, but it gave the same errors.  I am enclosing a file that shows the statement and all the errors.  I think I can used the code anyway, at least I'm moving on with it, but the Icon? type files still caused errors last evening.  Thanks!
[GitCloneErrors.txt](https://github.com/wroscoe/donkey/files/1844564/GitCloneErrors.txt)

wroscoe, 2018-03-24:
Ok. Just removed all those files. Didn't realize there were so many.

dschwee, 2018-03-24:
I cloned again in a different "test" directory, and this time, it completed with no errors:
(base) C:\Users\dschw\test>git clone https://github.com/wroscoe/donkey
Cloning into 'donkey'...
remote: Counting objects: 5142, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 5142 (delta 0), reused 1 (delta 0), pack-reused 5135
Receiving objects: 100% (5142/5142), 23.13 MiB | 908.00 KiB/s, done.
Resolving deltas: 100% (2889/2889), done.
Thank you!

closed 2018-03-24 by dschwee