autorope/donkeycar #230: Raspberry pi 3 B+ boot problem

onukadri, 2018-03-24:
Prebuilty zipped disk image is not work on Raspberry pi 3 B+. Flashes red light and wont boot.

wroscoe, 2018-03-24:
wow, you already have one. I'm working on updating the disk image and will post a script here.. https://github.com/wroscoe/donkey/blob/wr/disk_img_script/install/make_pi_disk_img.sh

onukadri, 2018-03-24:
I was waiting for this device and got one immediately. Thanks for fast
response.

onukadri, 2018-03-27:
hello again i have been getting this error "ImportError: numpy.core.multiarray failed to import", while using "donkey calibrate --channel 0" this line. i guess something wrong about numpy. i deleted and reinstalled numpy but no solution.

wroscoe, 2018-03-27:
https://stackoverflow.com/questions/20518632/importerror-numpy-core-multiarray-failed-to-import

try numpy 1.8?

onukadri, 2018-03-27:
https://stackoverflow.com/questions/17709641/valueerror-numpy-dtype-has-the-wrong-size-try-recompiling

Answer of RexFuzzle works for me. thanks

onukadri, 2018-03-27:
Now calibrate is working, my motor and servo are calibrated. however new error while "python manage.py drive" : 

tensorflow.python.framework.errors_impl.NotFoundError: /home/pi/env/lib/python3.5/site-packages/tensorflow/contrib/data/python/ops/../../_prefetching_ops.so: undefined symbol: _ZN6google8protobuf8internal26fixed_address_empty_stringB5cxx11E

wroscoe, 2018-03-27:
It would be helpful to see where in donkey that was getting called. On my Pi B the drive script is working. Haven't tried autopilots yet with this TF 1.5.

onukadri, 2018-03-27:
tensorflow 1.4.1 works well. thanks

MoreAJ, 2018-05-13:
Hey guys completely new to Raspberry pi's and donkeycars and im glad to have seen this issue posted. I am having the same issue where the prebuilt zipped disk image is not booting with our Raspberry pi so it is blinking red. Power supply is goood, Wroscoe where can i find a newer/updated version of this file? Or how can i do this properly? I am a complete noob!! Thank you

chinaq, 2018-05-16:
I am having the same issue.

meurig, 2018-05-21:
Just to say I was getting a similar but slightly different error:

`tensorflow.python.framework.errors_impl.NotFoundError: /home/pi/env/lib/python3.5/site-packages/tensorflow/contrib/data/python/ops/../../_dataset_ops.so: undefined symbol: _ZN6google8protobuf8internal9ArenaImpl10AddCleanupEPvPFvS3_E`

Following onukadri's suggestion of uninstalling tensorflow (I was on 1.7.0) and installing tesorflow 1.4.1 fixed it.

```
pip uninstall tensorflow
tf_file=tensorflow-1.4.1-cp35-none-linux_armv7l.whl
wget https://github.com/lhelontra/tensorflow-on-arm/releases/download/v1.4.1/${tf_file}
pip install ${tf_file}
rm ${tf_file}
```

I wasn't using an image, but instead installed by using this script as a guide:
https://github.com/wroscoe/donkey/blob/wr/disk_img_script/install/make_pi_disk_img.sh

scitronboy, 2018-05-22:
I used that script too. Almost ready to test, we'll see if it works.... I installed tensorflow 1.7 I think, on raspbian stretch, so I don't know if that will work.

meurig, 2018-05-22:
I’m on stretch lite if that helps

scitronboy, 2018-05-22:
It worked! I successfully installed opencv 3.4.1, tensorflow 1.4.1, and donkey on raspbian stretch.

wroscoe, 2018-07-31:
I've taken opencv out of the image. We're going pure tensor! :)  Here is the new install script... https://github.com/wroscoe/donkey/blob/dev/install/pi/install.sh

closed 2018-07-31 by wroscoe