autorope/donkeycar #242: trigger deploy to pypi using travis.ci

wroscoe, 2018-04-17:
Currently the install method is `git clone <repo>` but this uses the "default" branch witch  is current master. I'd like to make the default branch dev so this is what pull requests are made against. Here's a script to make deploying to pypi easy so the install setup could be `pip install donkeycar`

http://romgar.github.io/presentations/pypi_deployment/#slide10

joekickass, 2018-05-14:
I can have a look at this if you want

wroscoe, 2018-05-14:
This would be great. I Switched back to travis-ci in the dev branch because circle ci doesn't test pull requests.

Yes both assumptions are correct.  I'd like to deploy to pypi when a tag is made on the master branch.

wroscoe, 2018-05-28:
This now works. Thanks @joekickass

closed 2018-05-28 by wroscoe