autorope/donkeycar #250: Compability

MoreAJ, 2018-05-08:
Hey guys im new to Donkeycar and was hoping to get some feedback...i am aware of all the components needed to assemble this and would like to know if a Raspberry Pi 3 B+ will be compatible for doing this. I noticed some people use a Motherboard B or just a regular B model so i wanted to make sure a B+ will be fine. Please let me know!

MoreAJ, 2018-05-10:
Can anybody clarify this?

scitronboy, 2018-05-10:
Yes, donkey will work with a pi b+.

scitronboy, 2018-05-11:
See this commit: [Update the disk image to use tensorflow 1.7 and support pi3b+](https://github.com/wroscoe/donkey/pull/245)     Also, if your willing to put some work into it , you can make your own disk image with [this script](https://github.com/wroscoe/donkey/blob/wr/disk_img_script/install/make_pi_disk_img.sh). I had problems getting it working, mainly with compiling opencv, but I eventually got it.

MoreAJ, 2018-05-13:
thanks for the response scitronboy however, were having issues where we have been following the steps on donkeycar's website for setting up the Pi, and we cannot seem to find it on theWifi network on Ubuntu. We tried using the commands "findcar" "ping d2.local" "pingraspberrypi.local" in Ubuntu to try to locate the Raspberry pi. We did notice that the red light on the pi is flashing which i am not sure if it is supposed to. We loaded the SD card with the wifi text editor file containing our network name/pw and also completed the steps before it. Any idea as to why were having this issue? We are going to try plugging in a USB keyboard/mouse to see if it works. Let me know if you can give us some feedback! Thanks

scitronboy, 2018-05-13:
When you turn the pi on, you should see the green light turn in then start flickering. If you don't, it means the pi is not booting. If the green light flashes **solidly** a certain number of times then turns off, like on-off-on-off-on-off, it means that something is wrong with the sd card image. But, because of the red led flashing, it looks like you problem is power, not any of the above.

MoreAJ, 2018-05-13:
Oh i see that really explains alot...so it seems that the SD card image isnt working. Were complete noobs at this and currently trying to build this step by step. Im not sure where we went wrong but we followed the steps from Donkeycar's website as listed below: 

**"Get the Raspberry Pi working.
Before we can do anything we have to get our car's computer connected to the internet. The fastest way is to use the disk image created for donkey cars.

The method for using a disk image to create a bootable SD card varies between operating systems. These instructions are for Ubuntu but you can see more instructions here.

Download prebuilt zipped disk image (1.1GB).
Unzip the disk image.
Plug your SD card into your computer.
Open the "Startup Disk Creator" application.
Select your source disk image as the one you unzipped earlier.
Select your SD card as the disk to use.
Click "Make startup disk".
Setup the Pi's WiFi for first boot
We can create a special file which will be used to login to wifi on first boot. More reading here, but we will walk you through it.

On Windows, with your memory card image burned and memory disc still inserted, you should see two drives, which are actually two partitions on the mem disc. One is labeled boot. On Mac and Linux, you should also have access to the boot partition of the mem disc. This is formated with the common FAT type and is where we will edit some files to help it find and log-on to your wifi on it's first boot.

Start a text editor: gedit on Linux. Notepad on Windows. TextEdit on a Mac.
Paste and edit this contents to match your wifi:
country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="<your network name>"
    psk="<your password>"
}

Replace <your network name> with the ID of your network. Leave the quotes. I've seen problems when the network name contained an apostrophe, like "Joe's iPhone". Replace <your password> with your password, leaving it surrounded by quotes. If it bothers you to leave your password unencrypted, you may change the contents later once you've gotten the pi to boot and log-in.

Save this file to the root of boot partition with the filename wpa_supplicant.conf. On first boot, this file will be moved to /etc/wpa_supplicant/wpa_supplicant.conf where it may be edited later.
Setup Pi's Hostname
We can also setup the hostname so that your Pi easier to find once on the network. If yours is the only Pi on the network, then you can find it with

ping d2.local
once it's booted. If there are many other Pi's on the network, then this will have problems. If you are on a Linux machine, or are able to edit the UUID partition, then you can edit the /etc/hostname and /etc/hosts files now to make finding your pi on the network easier after boot. Edit those to replace raspberrypi with a name of your choosing. Use all lower case, no special characters, no hyphens, no underscores _.

sudo vi /media/userID/UUID/etc/hostname
sudo vi /media/userID/UUID/etc/hosts
Now you're SD card is ready. Eject it from your computer, put it in the Pi and plug in the Pi.**"

**So this is what we followed^ we took donkeys zipped disk image and extracted it then used Etcher to upload it onto the SD card. We then copied and pasted the network coding and created our own Network name and password. Im not sure what were doing wrong. Are we supposed to have Raspbian or Ubuntu downloaded onto the Pi board itself?

MoreAJ, 2018-05-14:
Id also like to note that we just bought the advised Anker battery on Donkeys website...so i do not think we should have issues with power anymore. I am starting to think it really is the disk image...is there an updated image somewhere??

wroscoe, 2018-05-14:
The disk image that came with my RPiB+ did the same thing and didn't work. I had to get a new SD card and make my own disk image to get around this power (?) issue.

closed 2018-05-14 by wroscoe