autorope/donkeycar #254: Donkeycar prebuild zipped disk image not booting with Raspberry Pi 3 B+

MoreAJ, 2018-05-13:
Hey guys hows it going...my friends and I are very new to using raspberry pi's and were trying to complete our donkey car. We have been following the Software instructions on donkeycars website but it seems that our B+ pi continues to blink red when attempting to boot it/setup wifi through our laptop.  It is definitely not a power supply issue because were using a 5V 2A battery. We unzipped the disk image and followed the steps as described using Ubuntu and uploaded the files onto our Sandisk SD card. As well we setup the network name and password and copied the text as described within the instructions. Is this zipped disk image out of date or not compatible with a Raspberry Pi 3 B+? How do we get our Pi to successfully get Wifi access? Are we supposed to have Raspian or NOOBS downloaded on the SD card? We have been just getting stressed out and feel like were running out of options to try. Please respond with some advice!

wroscoe, 2018-05-14:
Hey, This is a known issue and theirs a whole slack channel dedicated to it. However I've got it working using this script. https://github.com/wroscoe/donkey/blob/bd311a19acdc36b198882d3afbe9a023673ca007/install/make_pi_disk_img.sh

closed 2018-05-14 by wroscoe

MoreAJ, 2018-05-14:
Hey Wroscoe thanks for the response...however, i am just unsure how to properly do this with the use of the new script. Can you maybe explain the steps on how to do this? I am completely new with programming, raspberry pi's, and donkeycars. I had the understanding that i am supposed to use a zipped disk image? How can i learn to do this properly with the script to get the program working. Please let me know!

wroscoe, 2018-05-14:
> Appreciate it Will...so the way i would do this is in a Terminal window in
> Raspbian? Afterwards load this zipped disk image to the SD card then i
> should be able to find it on the Wifi network?

If you're going to use the disk image, you don't need the script. If you're
going to use the disk image you can follow the instructions on
docs.donkeycar.com

- Will