autorope/donkeycar #257: Cannot access Donkeycar directory or d2 for car calibrations

MoreAJ, 2018-05-15:
Hi guys im still very new to donkeycars and I am now having an issue where I cant seem to access the donkey directory after successfully connecting the IP address and through SSH? When we try to load up the donkey car directory, it says it does not exist. Wroscoe was nice to provide a zipped disk image for our Raspberry Pi 3 B+ however, we just cant seem to access the directory. After typing in the script 'nano ~/d2/config.py'  we are then brought to the GMU nano 2.7.4 with "Directory '~d2' does not exist)." We really do not know what is wrong as when we type in the following script 'git clone https://github.com/wroscoe/donkey.git donkeycar' we are left with 'fatal: destination path 'donkeycar' already exists and is not an empty directory.' My assumption would be that the donkey car directory and d2 should be accessible so that we may start calibrating our car. Can anyone please help out? Thank you

scitronboy, 2018-05-15:
If you used the prebuilt zipped disk image like you said you did, everything is already set up for you, so you do not need to type the command `git clone https://github.com/wroscoe/donkey.git donkeycar`. What this command does is download the donkeycar source code from github into the `donkeycar` directory, but the directory is already their on the prebuilt disk image so you get an error. What you need to do is update the source code, not download it. To do that, move into the source code directory with the command `cd ~/donkeycar`, and then type `git pull && sudo pip install -e`. This will update the source code. If you get any errors, let me know. After updating the source code, you must create a car directory. this directory will be the directory that all your cars data will be stored in - pictures it takes, calibration data, settings, models, etc.  To create this directory, simply type `donkey createcar --path ~/d2` then everything should work properly. Hope this helps!

everweb, 2018-05-16:
"Directory '~d2' does not exist" suggests that you miss-typed the 'nano ~/d2/config.py'.
Since '~' is your home directory, '~/d2' would be a 'd2' subdirectory under your home directory.
Try just using cd to change your current directory to '~/d2':
cd ~/d2
then you can use nano to edit the config.py file:
nano config.py

wroscoe, 2018-05-28:
You can always create a new "car app" with the command `donkey createcar --path ~/d2`

closed 2018-05-28 by wroscoe