autorope/donkeycar #260: Clean up dependency inconsistency

joekickass, 2018-05-17:
There are at least 4 _conda environment specification files_, and one **setup.py/setuptools** all listing a bunch of dependencies, of which some are inconsistent. A clean up would be nice, to

- *remove dependencies that are no longer needed or pulled in by other dependencies*, and
- *use the same library versions across different environments if possible*

Conda files:
- https://github.com/wroscoe/donkey/tree/master/envs

Setup.py
- https://github.com/wroscoe/donkey/blob/master/setup.py

There are additional dependencies on the [dev](https://github.com/wroscoe/donkey/tree/dev) branch.

tawnkramer, 2018-05-17:
some library versions were not available on different platforms at the time
this was created. You are welcome to test all these platforms and create a
pull request. But do no simply change them without testing, please.

joekickass, 2018-05-18:
I'll start and see if I can get far enough.

Should all libraries used in *donkeycar* be listed in *setup.py* or are there some exceptions? And should the conda environment files reflect those dependencies?

wroscoe, 2018-05-28:
I just removed keras from the env dependencies and upgraded the tensorflow versions. Donkeycar is also now installable from pypi. All the dependences should be listed in setup.py but the installation of the packages that require compilation should be completed outside of the pip workflow for the raspberry pi.

wroscoe, 2018-05-29:
Thinking more about this, we've got to have one way to install on each system, not many. For mac/windows/ubuntu this can be via conda. For raspberry pi I think the only reliable way to get all the dependencies will be via a disk image. This is unfortunate because it's a lot of work to prepare the disk image so it doesn't get update frequently. This may change if can get a whl built for tflite

joekickass, 2018-07-04:
@wroscoe - I was thinking it could be nice to have a trained model for every released donkey version to pypi. I'm planning to use it for both testing training functionality and backward compatibility. Do you have some good training data I can use? Also, do you have any preference on where to store binaries used for testing?

closed 2018-11-22 by wroscoe