autorope/donkeycar #271: Broadcast WIFI signal for connectivity

mancoast, 2018-06-25:
It would be useful if the system emitted its own local WIFI Access Point Host (hostapd) and DHCP server (dnsmasq).
REF:
https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

michaelbolanos, 2018-08-13:
I'm assuming it would be called  SSID:  donkeycar   Password?  

I've had this one one my radar, and I just did my first pull request today (ever) so if I can continue, and figure out how to contribute I will tackle this one.

Bleyddyn, 2018-08-13:
I've run mine both ways and find there are benefits to both. One thing I would strongly recommend is an easy, **reliable** way to switch from being a host to connecting to known hotspots. I still haven't gotten that to work. 

I've locked myself out of my robot several times trying to switch when I don't have an ethernet network handy.

wroscoe, 2018-11-22:
I think we absolutely need this and would be happy to work with anyone who wants to do this. The first step would be to write a spec for the proposed change, then we can have a google hangout to talk through the details. Maybe this needs a bounty?

mancoast, 2018-11-22:
I have already completed this on my build. I will post the instructions/requirements then implement.

mancoast, 2018-11-23:
Requires modification to the pre-built disk images to utilize WIFI Access Point Host (hostapd) and DHCP server (dnsmasq). REF: https://github.com/autorope/donkeycar/issues/348

Requires update to the documentation. REF: http://docs.donkeycar.com/guide/install_software/

tawnkramer, 2019-06-20:
do you have some instructions on how to do this?

mancoast, 2019-06-20:
Yea I can articulate the exacts when I find my notes on this project. In essence, I applied standards from Linux linked above to the raspberries. It’s only a few lines and I have it in a shell script somewhere. Works great for a classrooms where the internet WiFi will not allow remote control into the pi.

nvtkaszpir, 2019-11-09:
Any progress on this?
If not, I can add related docs how to do it.