autorope/donkeycar #272: Support WIFI network that does not have internet access

mancoast, 2018-06-25:
REF: https://github.com/wroscoe/donkey/blob/dev/donkeycar/util/web.py

The system determines its IP address using a ping to 8.8.8.8
This approach fails when the WIFI network does not have internet access.

michaelbolanos, 2018-06-25:
Hello William I've been trying to find a place to get started and try to help I've never done a pull request and I'm a bit trigger shy I would like to help on this request but I don't want to make an ass of myself and do things wrong.

If you can give me some pointers on how to do this the first time so I'm not asking too many questions or submitting noob questions I would be happy to try and assist on this one I've set up Linux Wireless in AP mode before.

mancoast, 2018-06-25:
sounds great, don't hesitate to get involved.

1.) modify the web.py file  https://github.com/wroscoe/donkey/blob/dev/donkeycar/util/web.py
2.) one approach is to utilize "try:" and "except:" statements
3.) return "127.0.0.1" #loopback when exception occurs

PR #273 (2018-06-27): support offline networks
resolves #272
closed 2018-06-27 by wroscoe