autorope/donkeycar #279: running manage.py drive get ImportError "No module named 'controller'"

Jokestv2, 2018-07-18:
I have update my git clone to the nearest version (v 2.5.1), and I have successfully "pip install -e ." and "pip install donkey[pi]", but still got this error.
By the way, I tried to mix it by changing "manage.py" file in mycar, instand of "import from controller", I set it to be "from donkeycar.parts.web_controller.web import LocalWebController".  This does fix the problem temporarily, but I can't record my data with the recording button pressed on on the website ( I do can control the car on the website).
Thanks a lot if anyone have an idea of this issue, this problem hinders three of my groups from processing on.
From: Beijing University of Posts and Telecommunications， China

tikurahul, 2018-07-31:
I think you are using the `dev` branch. Switch to `master` and run `pip install -e .` again. You should now be able to use the latest version.

igy254, 2018-08-01:
I think @tikurahul is right. When you clone donkey, git's HEAD points to dev branch.  Type `git branch` or `git branch -a` and you should see what branch you are using. Then if it is dev branch you are using, just type `git checkout master` in a directory where you have cloned donkey, and run `pip install -e .` in that same directory. Now it should be fine.

gang65, 2018-08-02:
@wroscoe Is that means that support for Joysticks were removed with commit: https://github.com/wroscoe/donkey/commit/311f999da904b17e3021f37e480fbf81bd77effe ?
According to documentation it is still there
https://docs.donkeycar.com/guide/get_driving/

I would recommend to not remove Joystick support, as it will allow to control vehicle without lag.

tikurahul, 2018-08-10:
@gang65 These commit only effects the `dev` branch. Please use the `master` branch which has the changes you need.

smahajan07, 2018-08-12:
I did checkout to master and did ```pip install -e .``` in the same directory , but I'm still getting the same error. 
```
~/mycar$python manage.py train --tub data/01/ model ./models/mypilot
using donkey v2.5.1 ...
Traceback (most recent call last):
  File "manage.py", line 24, in <module>
    from controller import LocalWebController, JoystickController
ImportError: No module named 'controller'
```

igy254, 2018-08-12:
@smahajan07 Have you activated donkey environment, before `pip install -e .`? By the way, you could copy your data to some other directory and run `donkey createcar ~/mycar`  again, but I don't see the reason for setup to not work if you have run the command inside donkey directory that you have cloned.

smahajan07, 2018-08-12:
So an important thing to mention is that after you install it again, you will have to run ```donkey createcar <name>``` again and then run the ```manage.py``` from it

smahajan07, 2018-08-12:
@igy254 I just saw your message! My bad! And to answer you, the environment was activated , I just had to run the ```donkey createcar ``` again. Thank you for your prompt response though!

Jokestv2, 2018-09-05:
@all Really sorry that I check this issue after so long, thanks a lot for your help, I will try it in no time.

frankthetank361, 2018-10-09:
I don't have a controller module either. I tried running "pip install -e", but it wants an argument (and I don't know what to put). I've also tried "donkey createcar", but it says, "createcar: error: the following arguments are required: path". Does anybody have a solution?

tikurahul, 2018-10-10:
You should do `pip install -e .` (You are missing a `.`).

tikurahul, 2018-10-11:
You will also need to rerun donkey create car with the right environment set. Make sure you read all the above comments.

frankthetank361, 2018-10-11:
Thanks, creating a new car solved the issue

yishuifengxing, 2018-11-08:
最新版本的donkey（2.5.1）为什么摇杆控制速度不可以了，也无法通过按键增加或减小速度。

scitronboy, 2018-11-09:
我正在使用谷歌翻译，所以请原谅这个中文的错误 I am using google translate so please excuse mistakes in this Chinese 我相信你所问的与这个帖子无关 I believe what you are asking doesn't relate to this post so please create a new issue here 所以，请 [在这里创建一个新问题。](https://github.com/autorope/donkeycar/issues/new)

closed 2018-11-22 by wroscoe