autorope/donkeycar #291: ModuleNotFoundError: No module named 'picamera'

miles-nash, 2018-08-18:
I have been working through installing the Donkey Car Software onto my Windows PC for the past few days and recently ran into an issue I cannot seem to sort out. After executing the 'python manage.py drive' command,  an error occurred and the trackback showed line 14 in ...\donkeycar\parts\camera.py with the error message "ModuleNotFoundError: No module named 'picamera'". I have repeatedly reinstalled the entirety of the Donkey Car Software onto my computer, but nothing seems to fix it. Any ideas as to what could be causing this?

alexa1287, 2018-08-18:
"python manage.py drive" command to be run on raspberry pi and not on PC

miles-nash, 2018-08-18:
After running 'manage.py' on my raspberry pi  I received another error message: 'picamera.exc.PiCameraError: Camera is not enabled.'. I went into raspi config and ensured that the camera was enabled, but after reboot, nothing happened.

miles-nash, 2018-08-18:
After running 'python manage.py drive',  an error occurs: "picamera.exc.PiCameraError: Camera is not enabled.". I have gone to raspi-config to enable the camera repeatedly, but the error persists. Thanks for any help.

closed 2018-08-19 by miles-nash

HarshNayal, 2021-10-02:
> "python manage.py drive" command to be run on raspberry pi and not on PC

so how can i use donkey car with python , any other way?