autorope/donkeycar #292: Problems with manage.py

miles-nash, 2018-08-19:
After executing 'python manage.py drive' this is what the terminal produced these messages.

'
/usr/lib/python3/dist-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
loading config file: /home/pi/mycar/config.py
config loaded
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/picamera/camera.py", line 456, in _init_camera
    self._camera = mo.MMALCamera()
  File "/home/pi/env/lib/python3.5/site-packages/picamera/mmalobj.py", line 2279, in __init__
    super(MMALCamera, self).__init__()
  File "/home/pi/env/lib/python3.5/site-packages/picamera/mmalobj.py", line 633, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/home/pi/env/lib/python3.5/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "manage.py", line 190, in <module>
    drive(cfg, model_path = args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 46, in drive
    cam = PiCamera(resolution=cfg.CAMERA_RESOLUTION)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/parts/camera.py", line 18, in __init__
    self.camera = PiCamera() #PiCamera gets resolution (height, width)
  File "/home/pi/env/lib/python3.5/site-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/home/pi/env/lib/python3.5/site-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
'

If anyone has any idea of how this happened and how to fix it it would be greatly appreciated.

alexa1287, 2018-08-20:
May be your camera is defective

distractedDave, 2018-08-24:
I see this both when the camera ribbon cable is backwards on one side, or when a ribbon cable gets broken.  Did you find the solution?

distractedDave, 2018-08-24:
Also this issue can happen if the screws on the camera are too tight and grounding out.

closed 2018-08-25 by miles-nash