autorope/donkeycar #301: Will not run according to training.

NikulausRui, 2018-08-31:
Thank you very much. After your guidance, I have solved many puzzles, but I have encountered new problems.

When I've finished a homemade track workout (driving 15 laps of data), use the trained pilots to make it drive automatically, it doesn't follow the training track, and it turns freely.

I trained in accordance with the documentation. Why did this happen?Can you give me some suggestions?

gang65, 2018-08-31:
Try gathering more data. I would recommend to drive in reverse direction. I think another 15 laps could be suffcient.

NikulausRui, 2018-08-31:
Thank you very much!I am testing it.

Another question is when I run python manage.py drive --model ./models/mypilot,and then choose LocalAngle,this mistake was occured:

Traceback (most recent call last):
  File "manage.py", line 190, in <module>
    drive(cfg, model_path = args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 131, in drive
    max_loop_count=cfg.MAX_LOOPS)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/vehicle.py", line 93, in start
    self.update_parts()
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/vehicle.py", line 129, in update_parts
    outputs = p.run(*inputs)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/parts/keras.py", line 75, in run
    angle_unbinned = util.data.linear_unbin(angle_binned[0])
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/util/data.py", line 38, in linear_unbin
    raise ValueError('Illegal array length, must be 15')
ValueError: Illegal array length, must be 15

Every time I can only run python manage.py and then choose LocalAngle,Will it be related to this?