autorope/donkeycar #317: OSError: [Errno 121] Remote I/O error

166inter, 2018-09-23:
Hello, I have been troubleshooting with an error for several hours now, and still not sure what is up.  I am trying to execute the `donkey calibrate --channel 0 ` command but am getting the i2c error in the image. I have tried swapping the motor shield, quadruple checking the connection, making sure i2c is enabled, re-flashing my sd, and still am encountering this. I have tried swapping my card into someone else's donkey car and was able to run and connect without any problem. 

[image]

Any ideas? I thing the problem might be hardware one.

rongfeng-china, 2019-03-21:
I have the same problem. It used to work well, but today it stops working. I've checked my i2c setting, and it's enabled already. Is this a hardware problem, and how to fix it? Thanks!

166inter, 2019-03-23:
@rongfeng-china
I still do not know what caused the error at that time, but more recently I encountered the i2c I/o error, and it turned out some of the pi GPIO pins were not sending a signal (from testing with Oscilloscope). If you can verify this, it may help narrow your problem. Cheers!