autorope/donkeycar #318: Not able to use a physical joystick along with web server

huberemanuel, 2018-10-01:
I'm not able to use my physical joystick with web server, but it works fine when I use the "--js" option, but that disables the web server, according to the docs: "The default web controller may be replaced with a one line change to use a physical joystick part for input ...", where is this line, and how do I change it?

paulo01012018, 2018-10-05:
That doesn't work with 2.5 and above, only with 2.4

scitronboy, 2018-10-15:
> "The default web controller may be replaced with a one line change to use a physical joystick part for input ...", where is this line, and how do I change it?

I think that line might be in config.py:

    USE_JOYSTICK_AS_DEFAULT = False

maybe you need to change it to:

    USE_JOYSTICK_AS_DEFAULT = True

scitronboy, 2018-10-16:
Yes, but have you tried starting the web server with `USE_JOYSTICK_AS_DEFAULT = True` enabled in the config?

huberemanuel, 2018-10-16:
Yeap, look at that code:
`    if use_joystick or cfg.USE_JOYSTICK_AS_DEFAULT:
         ctr = JoystickController(max_throttle=cfg.JOYSTICK_MAX_THROTTLE,
                                  steering_scale=cfg.JOYSTICK_STEERING_SCALE,
                                  auto_record_on_throttle=cfg.AUTO_RECORD_ON_    THROTTLE)
     else:
         # This web controller will create a web server that is capable
         # of managing steering, throttle, and modes, and more.
         ctr = LocalWebController(use_chaos=use_chaos)`
If you send the option '--js' or set in the config file `USE_JOYSTICK_AS_DEFAULT = True` it won't start the web server.

scitronboy, 2018-10-16:
I see...

What's wrong with your "temporary fix"? Why don't you create a PR for it?

huberemanuel, 2018-10-17:
Because it wasn't made to work in that way, a complete solution would involve changing the donkey architecture, that I may try in the future.

scitronboy, 2018-10-17:
Ok.

holysh1t, 2018-11-07:
What does '...disables the Webserver.' mean? Does it say 'device /dev/jd0' not found? or what?

scitronboy, 2018-11-07:
What?

holysh1t, 2018-11-07:
> What?

@scitronboy I did mean OP.

scitronboy, 2018-11-08:
Sorry @holysh1t  I misunderstood your question. "Disables the web server" means that donkey doesn't serve the web server (for you to drive your car with) to http://localhost:8887/

holysh1t, 2018-11-08:
@scitronboy No problem at all. Yes I also know that, but why doesn't it serve the webserver, there must be some sort of error message?

holysh1t, 2018-11-08:
@huberemanuel That can't be true, if I add the '--js' option it says "[...] device /dev/jd0 not found [...]"! I dont think that adding the parameter just stops the script from working.

scitronboy, 2018-11-09:
@holysh1t 

> it says "[...] device /dev/jd0 not found [...]"!

I think that's just because the pi can't find the controller. If you plug in a valid joystick, the script should run normally without an error, it just won't start the webserver (because that's how it's designed in the code - _not_ because of an error).

huberemanuel, 2018-11-09:
@holysh1t what @scitronboy said is true, this isn't an error, just how the app is designed, but it sucks.

holysh1t, 2018-11-09:
@huberemanuel I missunderstood, sorry. That indeed sucks.

wroscoe, 2018-11-11:
This has a work around as described above. I also wrote an alternative part for the Wii u controller.... https://github.com/autorope/donkeypart_bluetooth_game_controller

closed 2018-11-11 by wroscoe