autorope/donkeycar #322: Sanitizing User Input when Calibrating: "ValueError: invalid literal for int() with base 10"

rampageservices, 2018-10-13:
Hello,

I use a ssh program that works in powershell installed via www.scoop.sh.
This does not happen in OSX terminal ssh.

I run into a ValueError whenever I press the backspace during calibration.
While my situation is a edge case, the bigger issue is that It appears there is no sanitation done on user input for this step of the process.

The code in question is the following:
`int(input('Enter a PWM setting to test(0-1500)'))`

My computer science professors used to call this sloppy ninja code. You are casting user input as an int without checking the value first. At the very least, the input should be checked to see if it is within bounds, 0-1500. 

In the best case scenario, only digits should be assigned to the pwm variable.

See this stack overflow question for a few basic examples:
https://stackoverflow.com/questions/17336943/removing-non-numeric-characters-from-a-string

Thanks!

`(env) pi@donkeypi:~/mycar $ donkey calibrate --channel 0
using donkey v2.5.1 ...
Enter a PWM setting to test(0-1500)401
Enter a PWM setting to test(0-1500)402
Enter a PWM setting to test(0-1500)403
Enter a PWM setting to test(0-1500)4034^H^H4
Traceback (most recent call last):
  File "/usr/local/bin/donkey", line 11, in <module>
    sys.exit(execute_from_command_line())
  File "/usr/local/lib/python3.5/dist-packages/donkeycar/management/base.py", line 445, in execute_from_command_line
    c.run(args[2:])
  File "/usr/local/lib/python3.5/dist-packages/donkeycar/management/base.py", line 156, in run
    pmw = int(input('Enter a PWM setting to test(0-1500)'))
ValueError: invalid literal for int() with base 10: '4034\x08\x084'`

scitronboy, 2018-10-15:
Also, In my opinion, people should be able to read the prompt and enter a number between 0-1500. If they don't, I think python's error message is good enough to fix the problem.

rampageservices, 2018-10-15:
Sure, I can invest the time to fix it and submit a pull request.
However, the point of creating an issue is to discuss the issue before any action is taken.
Maybe, someone has already written their own fix, maybe someone feels strongly that they want to submit the pull request. 

Further, you are entitled to your opinion but I don't agree with your sentiment.

If Donkey car wants to see new user adoption the essential function of calibration needs to be bulletproof. Otherwise, users will get frustrated and no longer work on the project having never even gotten to the fun part of actually seeing the car drive itself.

Further, the error message is nonsense to any non programmer.

I am a strong believer in failing gracefully instead of failing hard. 
In this case, this is a hard failure that should be remediated.

scitronboy, 2018-10-16:
#325

rampageservices, 2018-11-26:
This issue was closed hastily. A better solution was proposed and was not discussed.