autorope/donkeycar #328: FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1

kelvinkoko, 2018-10-19:
When I start to run `python manage.py drive` I hit the following error  (boot from pi zero image)
```
(env) pi@donkeypi:~/mycar $ python manage.py drive
using donkey v2.5.1 ...
/usr/lib/python3/dist-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
loading config file: /home/pi/mycar/config.py
config loaded
PiCamera loaded.. .warming camera
Starting Donkey Server...
Traceback (most recent call last):
  File "manage.py", line 190, in <module>
    drive(cfg, model_path = args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 103, in drive
    steering_controller = PCA9685(cfg.STEERING_CHANNEL)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/parts/actuator.py", line 18, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685()
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PCA9685/PCA9685.py", line 74, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_GPIO/I2C.py", line 64, in get_i2c_device
    return Device(address, busnum, i2c_interface, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_GPIO/I2C.py", line 97, in __init__
    self._bus = Adafruit_PureIO.smbus.SMBus(busnum)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 105, in __init__
    self.open(bus)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 130, in open
    self._device = open('/dev/i2c-{0}'.format(bus), 'r+b', buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'
```
But it can be easily fixed by 
adding (or uncomment) `dtparam=i2c_arm=on`
in `/boot/config.txt`

Just would like to see is it a common issue

scitronboy, 2018-10-19:
Hello, it states in the donkey instructions that you must enable i2c in `raspi-config`.  Did you do that?

kelvinkoko, 2018-10-19:
@scitronboy Ah, I haven't done that. Thanks for the prompt reply!~ I found it in the faq now. 
Originally I am following this: http://docs.donkeycar.com/guide/install_software/
But that part seems just mention to use `raspi-config` to expand file system. Do you think it is good to mention it is necessary to enable i2c in that page?

scitronboy, 2018-10-19:
I don't think it's necessary on that page, because the pre-built disk image should already have i2c enabled. Are you using the pre-built disk image? Also, try checking raspi-config: `sudo raspi-config` and then navigate to `interfacing options` and then check i2c. Is it enabled???

kelvinkoko, 2018-10-21:
Yes, i am using the pre-build image (for pi-zero) but i agree user can enable it in raspi-config easily even not enabled. let me close this issue

closed 2018-10-21 by kelvinkoko