autorope/donkeycar #339: Donkey Sim command cannot find the name

jaskarannagi19, 2018-11-05:
Donkey sim

using donkey version: 2.5.7 ...
Traceback (most recent call last):
  File "/home/jaskaran/.conda/envs/DonkeyCar/bin/donkey", line 11, in <module>
    load_entry_point('donkeycar', 'console_scripts', 'donkey')()
  File "/home/jaskaran/donkeycar/donkeycar/management/base.py", line 456, in execute_from_command_line
    c.run(args[2:])
  File "/home/jaskaran/donkeycar/donkeycar/management/base.py", line 258, in run
    from donkeycar.parts.simulation import SteeringServer
ImportError: cannot import name 'SteeringServer'

Any help here

scitronboy, 2018-11-05:
When did you install donkey? @wroscoe might this be a problem caused by #338 ?

jaskarannagi19, 2018-11-05:
@scitronboy I installed it 6-7 hours ago

jaskarannagi19, 2018-11-05:
@scitronboy ok I'll try with older commit than and will let you know in sometime

wroscoe, 2018-11-05:
Try installing version 2.5.1

jaskarannagi19, 2018-11-05:
@wroscoe This did help that error is gone

closed 2018-11-05 by jaskarannagi19

macfly1202, 2019-02-11:
just install today donkey and see exact same error:
<code>(car) jmp@deep:~/mycar$ donkey sim --model=models/mypilot
using donkey version: 2.5.7 ...
Traceback (most recent call last):
  File "/home/jmp/anaconda3/bin/donkey", line 9, in <module>
    load_entry_point('donkeycar==2.5.7', 'console_scripts', 'donkey')()
  File "/home/jmp/Dropbox/github/donkeycar/car/donkeycar/donkeycar/management/base.py", line 457, in execute_from_command_line
    c.run(args[2:])
  File "/home/jmp/Dropbox/github/donkeycar/car/donkeycar/donkeycar/management/base.py", line 259, in run
    from donkeycar.parts.simulation import SteeringServer
ImportError: cannot import name 'SteeringServer'
</code>

Will try to install previous donkey version...