autorope/donkeycar #347: Xbox One Controller for steering

niallmandal, 2018-11-21:
Does anyone know how to set up an Xbox One wireless controller to steer the donkey car while collecting training data? I am looking to fine tune manual steering. I have seen many ways to do this, I am just not sure how.  Do you know of any ways to do this directly, or as an add-on through the browser interacting with elements on the page (i.e. [gamepad](chrome://flags/#enable-gamepad-extensions))?

wroscoe, 2018-11-21:
If this is a bluetooth controller you could try [this package](https://github.com/autorope/donkeypart_bluetooth_game_controller) to add it as a controller to your donkeycar. You'll probably need to create an updated config file that gives the button mapping for the xbox. I wrote this package for the WiiU.

The default config file: https://github.com/autorope/donkeypart_bluetooth_game_controller/blob/master/donkeypart_bluetooth_game_controller/wiiu_config.yml

closed 2018-11-21 by wroscoe

wroscoe, 2018-11-21:
Updated the readme to show how to add support for a new joystick. https://github.com/autorope/donkeypart_bluetooth_game_controller/blob/master/README.md

niallmandal, 2018-11-21:
The link to the Wii U config file is invalid. Can you please fix this?

niallmandal, 2018-11-21:
can you explain how the config file works,  and how I should tackle editing it, please?

wroscoe, 2018-11-21:
When you have your joystick connected vai bluetooth, running the part.py script will print out the buttons and their codes as you press them. For example with the WiiU controller when you press the bottom left trigger the code 312 will show. If you see a different code you'll need to change that code that's mapped to that button to whatever code you see.

niallmandal, 2018-11-21:
When running part.py, an error is raised that looks like this

`FileNotFoundError: [Errno 2] No such file or directory: './donkeypart_bluetooth_game_controller/donkeyblue/wiiu_config.yml'`

Doesn't this mean that part.py is relying on a file that I am supposed to create?
(I switched to testing with a regular Wii remote and Nunchuk, but the error is still present)

niallmandal, 2018-11-21:
Not to mention, runnning part.py requires that the device is named "nintendo." This is why I had to switch to the Wii Controller. Am I doing something wrong?

wroscoe, 2018-11-22:
No, you're just the first person to add a new controller. I've pushed some updates that will ask you for a search term to find your controllers input device. I also added the wiiu_config.yml file for installs done without the -e flag.

niallmandal, 2018-11-22:
when do you think these updates will be done
.
.
.
sorry for the rush i’m building the car for a competition and training needs to happens soon :)

wroscoe, 2018-11-22:
What updates? You should have what you need to use it for a different BT controller. If its not related to this issues you can msg me on slack.

niallmandal, 2018-11-22:
the updates to the code. you said you pushed the updates. when will these updates be added? Or have they been already added?

niallmandal, 2018-11-22:
ohhh okay thank you! i will contact you over slack if i have any more questions or errors. Thank you so much

niallmandal, 2018-11-22:
So I ran the new files with a wii remote, and the same error is still here. What should I do?
[image]
[image]