autorope/donkeycar #353: ValueError: Unknown initializer: GlorotUniform

aldajo92, 2018-11-29:
When I pass a trained model made in my computer and sent to the Rpi, I get the following error after pass the comand ```python manage.py drive --model ~/mycar/models/mypilot``` :

```Traceback (most recent call last):
  File "manage.py", line 176, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 77, in drive
    kl.load(model_path)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/parts/keras.py", line 19, in load
    self.model = load_model(model_path)
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/engine/saving.py", line 241, in load_model
    model = model_from_config(model_config, custom_objects=custom_objects)
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/engine/saving.py", line 318, in model_from_config
    return deserialize(config, custom_objects=custom_objects)
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/layers/serialization.py", line 63, in deserialize
    printable_module_name='layer')
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/utils/generic_utils.py", line 171, in deserialize_keras_object
    list(custom_objects.items())))
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/engine/network.py", line 1057, in from_config
    process_layer(layer_data)
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/engine/network.py", line 1043, in process_layer
    layer = deserialize_layer(layer_data, custom_objects=custom_objects)
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/layers/serialization.py", line 63, in deserialize
    printable_module_name='layer')
  File "/home/pi/env/lib/python3.5/site-packages/tensorflow/python/keras/_impl/keras/utils/generic_utils.py", line 173, in deserialize_keras_object
    return cls.from_config(config['config'])
ValueError: Unknown initializer: GlorotUniform```

rychitre, 2018-12-04:
I am getting exact same error. Did you find solution?

rychitre, 2018-12-04:
What tensorflow and keras version are you usong on machine and Raspberry Pi?

yishuifengxing, 2018-12-06:
My version of TensorFlow is 1.11.0

yishuifengxing, 2018-12-06:
> What tensorflow and keras version are you usong on machine and Raspberry Pi?

My version of TensorFlow is 1.11.0

rychitre, 2018-12-07:
I was able to resolve the issue. Tensoflow version on Rpi is 1.8. You need to install same version on computer. or you can train model directly in Rpi.

aldajo92, 2018-12-07:
This error for me was caused with tensorflow 1.8 in both, Rpi and computer.

yishuifengxing, 2018-12-08:
> I was able to resolve the issue. Tensoflow version on Rpi is 1.8. You need to install same version on computer. or you can train model directly in Rpi.

TensorFlow is a GPU version of my computer. When I used donkey-2.5.1, this problem did not arise. If you use raspberry pie for a long time.

yishuifengxing, 2018-12-09:
I know the problem is that my computer TensorFlow version was 12.0, while the raspberry pie version was 11.0. I changed the TensorFlow version of my computer to 11.0 and there was no problem.

claick-oliveira, 2018-12-29:
Hi guys, I've changed a few things and solved this problem. Try this! In my situation I used version 1.10.1 of the tensorflow and tensorflow-gpu

https://github.com/claick-oliveira/donkeycar/tree/fix_and_changes

benjaminpochat, 2019-03-28:
I had the same problem with a model built with tensorflow 11.1.0 (using tensorflow.python.keras.models.save_model) and loaded with tensoflow 1.11.0 (using tensorflow.python.keras.models.load_model).

I solved it by upgrading everything to tensorflow 1.13.1, after building the model again with the new version, I could load it without this error.

tawnkramer, 2019-06-20:
should be fixed now. please update and try.

closed 2019-06-20 by tawnkramer