autorope/donkeycar #356: how to drive can on its own without using web/phone

rychitre, 2018-12-07:
I trained the model and ran using ‘python manage.py drive --model ~/mycar/models/mypilot’.  When I run this command it gives me below. So, basically, it is asking me to drive the car manually using phone or website. 

So, this is manual driving. How can car drive by itself? What should be done to drive on its own?

--------------------------------------------------------------------------------------------------------------------------------------------------------------------
(env) pi@donkeypi:~/mycar $ python manage.py drive --model ~/mycar/models/mypilot
using donkey version: 2.5.8 ...
/usr/lib/python3/dist-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
loading config file: /home/pi/mycar/config.py
config loaded
PiCamera loaded.. .warming camera
Starting Donkey Server...
You can now go to http://10.x.x.xx:8887 to drive your car.
/home/pi/env/lib/python3.5/site-packages/picamera/encoders.py:544: PiCameraResolutionRounded: frame size rounded up from 160x120 to 160x128  width, height, fwidth, fheight)))
WARNING:tornado.access:404 GET /favicon.ico (10.0.0.15) 12.18ms
--------------------------------------------------------------------------------------------------------------------------------------------------------------------

Schmill, 2019-05-30:
You need to navigate to the mentioned webpage, then select the correct pilot mode from there.

Taken from the "Train an autopilot" part of the documentation:

> Mode & Pilot: Congratulations on getting it this far. The first thing to note after running the command above, is to look at the options in the Mode & Pilot menu. It can be pretty confusing. So here's what the different options mean:
> 
> a. User : As you guessed, this is where you are in control of both the steering and throttle control.
> 
> b. Local Angle : Not too obvious, but this is where the trained model (mypilot from above) controls the steering. The Local refers to the trained model which is locally hosted on the raspberry-pi.
> 
> c. Local Pilot : This is where the trained model (mypilot) assumes control of both the steering and the throttle. As of now, it's purportedly not very reliable.

tawnkramer, 2019-06-20:
please re-open if still an issue.

closed 2019-06-20 by tawnkramer