autorope/donkeycar #357: Error when running `python manage.py drive`

jamestharpe, 2018-12-09:
I'm getting `ImportError: No module named 'donkeycar.parts.controller'` when I run `python manage.py drive` according to the [calibration instructions](http://docs.donkeycar.com/guide/calibrate/).

```bash
(env) pi@donkeypi:~/mycar $ python manage.py drive
using donkey version: 2.5.7 ...
/usr/lib/python3/dist-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
Traceback (most recent call last):
  File "manage.py", line 27, in <module>
    from donkeycar.parts.controller import LocalWebController, JoystickController
ImportError: No module named 'donkeycar.parts.controller'
```

I'm not familiar with Python. Do I need to install something else?

jamestharpe, 2018-12-10:
Checking out tag 2.5.6 fixes this. Looks like 2.5.7 is not quite ready, or the instructions need to be updated. Not sure which.

scitronboy, 2019-02-02:
@tikurahul This _is_ the autorope repo.

tikurahul, 2019-02-02:
There are other github repos which have the parts that have been separated from the main repo.

tawnkramer, 2019-06-20:
should be fixed now.

closed 2019-06-20 by tawnkramer