autorope/donkeycar #360: NameError: name 'JoystickController' is not defined

custom-build-robots, 2018-12-09:
Hello everyone,
since I installed the latest donkey car version 2.5.8 I am no longer able to use my PS4 controller connected via DualShock 4 USB Wireless Adaptor. In the past I had no problem to use the PS4 controller to steer my Donkey Car.

Below is the error I get since I updated today my Donkey Car. I downloaded the latest image for the Raspberry Pi 3 Model B+ and installed verything as described.

ERROR MESSAGE:
(env) pi@donkeypi:~/mycar $ python manage.py drive
using donkey version: 2.5.8 ...
/usr/lib/python3/dist-packages/h5py/__init__.py:34: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
loading config file: /home/pi/mycar/config.py
config loaded
PiCamera loaded.. .warming camera
Traceback (most recent call last):
  File "manage.py", line 176, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 48, in drive
    ctr = JoystickController(max_throttle=cfg.JOYSTICK_MAX_THROTTLE,
NameError: name 'JoystickController' is not defined

This picture shows the donkey car error message I now get:
[image]

If I start this little python program to test my PS4 controller it works perfct:

#!/usr/bin/env python
import struct
infile_path = "/dev/input/js0"
EVENT_SIZE = struct.calcsize("LhBB")
file = open(infile_path, "rb")
event = file.read(EVENT_SIZE)
while event:
    print(struct.unpack("LhBB", event))
    (tv_msec,  value, type, number) = struct.unpack("LhBB", event)
    event = file.read(EVENT_SIZE)

This picture shows the output of my python joystick test program:
[image]

Thank you very much for your help or ideas

Ingmar

ndnparis, 2018-12-16:
Hi,
I have the same issues with my PS3 joystick.
It works with other apps but not with donkeycar.
Hope someone could help us.

kevkruemp, 2019-01-19:
The newest update does appear to be missing the proper controller file. I will see if this can be remedied.

vihank, 2019-02-17:
> The newest update does appear to be missing the proper controller file. I will see if this can be remedied.

Has this issue been fixed yet?

jmonsalve15, 2019-04-07:
Hi @custom-build-robots, I am having the same issue but I don’t have the old versions of Donkey repo, can you share the missing file or show me from where could I download? I am pretty new and I getting struggles to found info here. Thanks

pitkane, 2019-05-02:
As Ingmar said, copying the file `controller.py` from older build kind of solves the problem.

```
cd ~/env/lib/python3.5/site-packages/donkeycar/parts/
wget https://github.com/autorope/donkeycar/blob/2.5.1/donkeycar/parts/controller.py

```

In `manage.py`
```
from donkeycar.parts.controller import JoystickController
```

tawnkramer, 2019-06-11:
please be on the lookout for donkey 3.0. This will have all the controller support in the main repo.

closed 2019-06-11 by tawnkramer