autorope/donkeycar #363: Website install instructions for Mac installs wrong version of tensorflow

Ezward, 2018-12-09:
The website says to specifically install tensorflow 1.3 in the Mac instructions here: http://docs.donkeycar.com/guide/install_software/#install-donkeycar-on-mac

When running manage.py train, that results in an error:
ModuleNotFoundError: No module named 'tensorflow.python.keras'

I've been using those instructions; are there alternate instructions I should be using?

bahree, 2019-03-11:
I am curious - which version version of TensorFlow did you install in the end? I see the documentation now explicitly says tensorflow-1.10.1. But that later causes issues with MoviePy. Thanks in advance.

dcremonini, 2019-03-14:
I found that in file
/docs/guide/install_software.md
in the dev and master branch there are some differences about the version (among others):
For example
in the dev branch there is no pip install tensorflow==1.8.0
while in the master branch I see at the line 163: pip install tensorflow==1.8.0
This is something that should probably be amended.

Ezward, 2019-03-15:
@bahree I did not explicitly install tensorflow at all; it get's installed by ananconda with the donkey environment.  I noticed this because the console logging while installing the version in the instructions showed that in actually uninstalled a version of tensorflow, then installed the 'wrong' version.

tawnkramer, 2019-06-20:
should be fixed now.

closed 2019-06-20 by tawnkramer