autorope/donkeycar #365: python manage.py drive remote IO error

xzho2604, 2019-01-05:
Hi, 
when I run python manage.py drive 
I got the error below:

`Traceback (most recent call last):
  File "manage.py", line 176, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'], use_chaos=args['--chaos'])
  File "manage.py", line 103, in drive
    steering_controller = PCA9685(cfg.STEERING_CHANNEL)
  File "/home/pi/env/lib/python3.5/site-packages/donkeycar/parts/actuator.py", line 19, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685()
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/pi/env/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 256, in write_byte_data
    self._device.write(data)
OSError: [Errno 121] Remote I/O error`

Would appreciate if some one would help!
Cheers
Erik

scitronboy, 2019-01-05:
Is I2C enabled in the interfacing options in `raspi-config`?

xzho2604, 2019-01-14:
Hi scitronboy,

Thank you for your reply, I have enable I2C and the pi2 camera works but the wide angle camera from the donkey store would not work......

Thank you
Erik

tawnkramer, 2019-06-12:
Here's some tips on solving that error:
https://donkey.discourse.group/t/faq-troubleshooting/33/3?u=tawnkramer

closed 2019-06-12 by tawnkramer