autorope/donkeycar #366: Questions about different software and hardware

filips123, 2019-01-09:
I want to use Donkeycar with my own car which uses different hardware that the official one.

I have few questions about that.

---

I use L298N motor driver. My parts are similar to that in [Raspberry Pi guide](https://projects.raspberrypi.org/en/projects/build-a-buggy), but I also have normal (not wide angle) Raspberry Pi camera.

Will that work? What I need to change to work?

---

To what angle should camera be tilted? What distance should be between car and nearest point in camera view? See bellow image for what I mean.

[image]

---

Is it possible to control car with my own software?

I want to have different website which will have support to control car, start and stop data recording (with also displaying how many frames and seconds is recording long) and start and stop drive. I also want to have buttons to transfer data and models between RPi and computer.

---

What data needs to be collected for training and in what format? Should just images be collected or should it also collect other data?

Can I make my own program that will collect data in required format for training?

---

My camera and motors are controlled in some specific way.

Is it possible that Donkeycar just get images from camera (in MJPEG stream or frame-by-frame stream) and just outputs in which directions should motors go? I want to control them in different ways.

scitronboy, 2019-01-09:
Anything is possible! Yes, you can control the car with your own software, but obviously if you want to work with donkey as well as your own software you'll need to modify donkey to make it work with your software. You can use a different website/interface, but you're going to have to program it yourself. 

In training, donkey collects images and throttle/steering data in the form of one JSON file for each frame. Yes, you can make your own training system, but, again, you're going to have to program it yourself. Same for all the rest of your question.

It is possible to detect traffic lights (for example with OpenCV); and other people have done it; but that functionality is not integrated into donkey right now. If you develop a good way of doing this on the pi feel free to submit a pull request.

I suggest that you look through all the code yourself and try to figure out how it works. Also I suggest that you build a donkey using the default setup and code _first_ before you start trying to make modifications. Ask if you have any questions.

filips123, 2019-01-09:
What about camera (see my original question)? Also, is it needed to change it's focus?

What are that throttle/steering data? How can I collect them myself?

Also, can you point me a bit where in the code are what things and what should I change?

scitronboy, 2019-01-09:
I don't know about the camera - you could probably just try yours.

As I already said, you should start by using the default donkey data collector.

What code do you want to change???

filips123, 2019-01-10:
I mean what code should be changed or added by me to integrate my website and programs with donkey. What functions should I call, where are they located...

scitronboy, 2019-01-10:
Start by watching the videos by @tawnkramer  that explain the donkey car parts system.

[Here is the first one.](https://www.youtube.com/watch?v=YZ4ESrtfShs)

filips123, 2019-01-10:
Thank you!

What are that throttle/steering data? How can I collect them myself?
From [the docs](http://docs.donkeycar.com/guide/calibrate/) I see that they are for ESC motors. I'm using different motors. Do I also need them?

scitronboy, 2019-01-10:
Once you actually build a car using the official instructions you can look at the data and figure the structure out. Then you will know how you need to store the data. The [code that stores the data](https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/datastore.py) is here.

You need the throttle-steering data always;it does not matter which car you use. If you are not using the standard car you will have to develop your own code to interface it to the donkey software. Many people have done this before (including me). What type of car do you have??? Do you already have a motor driver for it?

Also keep in mind that you can ask questions like this on the [slack channel](http://www.donkeycar.com/community.html).

filips123, 2019-01-11:
I will build my own custom car. I will use L298N motor driver.

closed 2019-06-12 by tawnkramer

filips123, 2019-06-12:
Thanks for closing this.
I can confirm that the car also works with other motors. I just had to update motor controllers and some other things.