autorope/donkeycar #369: Power Raspberry Pi and Servo motor with one source

mahmoud891, 2019-01-24:
Hello everyone , i am trying to create my own Car , following the documentation part on how to do it , as i was preparing the Hardware part , i saw that there is one problem , there is written " do not connect Servo Motor directly to the 5v pin of Raspberry Pi because it is not safe " and instead they are using Power Bank ( Anker ) for powering the Raspberry Pi and Servo together . My question is : Would it be a problem if we power Raspberry Pi with Power Bank which has 5 Volts , and we connect Servo motor 4,8-6 V through PCA9685 to Raspberry Pi ? if Servo motor is 4,8 - 6 Volt and we supply it with Vin=5V there is 1 volt difference , is it dangerous ? Thank you in advance .

DocGarbanzo, 2019-02-12:
Hi Mahmoud,
The servo usually gets powered through the ESC which is connected to the PCA9685. It has an onboard voltage regulator (called BEC) providing the right voltage- I believe this is 5V -  and enough current (1A) to do so. The ESC is powered through your battery which is probably a 2S Lipo or similar. With a cheap small buck converter / voltage regulator you can tap into your battery and power the Pi. I am using https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F163095874791, I believe there are many options. This one allows for 3A, which looks to be on the safe side.

tawnkramer, 2019-06-12:
The Sombrero hat available in the donkeycar store makes this easy.

closed 2019-06-12 by tawnkramer